ouster-lidar/ouster-ros #322: Show low frequency when recording rosbag or echoing /ouster/points topic & Artifact of ROS driver

SangHunHan92, 2024-04-17:
**1. Show low frequency when recording rosbag or echoing /ouster/points topic with ROS2 galactic driver**
When I launch ouster driver with ```ros2 launch ouster_ros driver.launch.py``, lidar pointcloud showing with ```rviz2``` shows a sufficiently high frequency (probably 20hz) [Fig 1].
[image]

However, if I record or echo ```/ouster/points``` topic, lidar frequency is significantly slower. [Fig 2]
When I stopped record or echo, the original frequency is restored.
[image]

Surprisingly, ROS noetic driver doesn't have this problem at all. Both record and echo work well at 20hz.

During this process, I checked the data transfer amount with ```iftop```, but there was no change. (Constant at about 60Mb)
Changing resolution of the pointcloud also doesn't work: even the lowest 512x10 doesn't solve the issue.
What's the solution for ROS2 galactic?

**2. Show low frequency when recording rosbag or echoing /ouster/points topic with ROS2 galactic driver**
When I launch ouster driver, lidar pointcloud in ```rviz``` show line artifact [Fig 3]. This issue shows either ROS and ROS2 driver.

[image]
[image]
(Examples of ROS noetic)

[image]
(Example of ROS2 galactic)

But, this artifact doesn't show in OusterStudio [Fig 4]. 
[image]

**Platform (please complete the following information):**
- Ouster Sensor? OS-1-32-U
- Ouster Firmware Version? 2.5
- ROS version/distro? ros2 galactic, ros noetic
- Operating System? Linux ubuntu 20.04 focal
- Machine Architecture? amd64
- git commit hash (if not the latest).

andre-nguyen, 2024-04-17:
Is this possibly the same issue as #240 and #272 ? Have you tried https://github.com/ouster-lidar/ouster-ros/pull/302 or #320 ?

Samahu, 2024-04-17:
@SangHunHan92 what @andre-nguyen said, most likely it is the same problem. For ROS[noetic] if you can pull #319 and see if that helps your situation or wait till it gets merged.

Samahu, 2024-04-17:
I just noticed that you mentioned you were under ROS[galactic] so the PR that @andre-nguyen referenced is the right one