ouster-lidar/ouster-ros #325: Trouble with launching the nodes through driver.launch and sensor.launch

SoroushMalekiXP, 2024-05-09:
**Describe your question**
I would like to test our Ouster LiDAR on Ubuntu 18 ROS Melodic and I have been facing difficulties with the driver. Before anything, I am aware that the support for Melodic has been dropped, however, due to other dependencies I have to use Melodic. At first, I was running into the following error when trying to build the project, when trying exactly as described in the Readme file.
`
/home/soroushmaleki/catkin_ws/src/ouster-ros/ouster-sdk/ouster_client/src/curl_client.h:95:43: error: no matching function for call to 'spdlog::logger::warn(std::__cxx11::basic_string<char>, const string&, long int&, std::__cxx11::string&)'
                     url, http_code, buffer);` 

Since this is just a problem with the Logger, I simply commented this line of code and were able to build the driver. However, upon trying to launch the driver.launch file using the following command:

`roslaunch ouster_ros driver.launch sensor_hostname:=os1-122023000117`

I run into the following error:

```
`... logging to /home/soroushmaleki/.ros/log/bbf0b0f4-0e2c-11ef-9c13-7c10c952880f/roslaunch-soroushmaleki-ROG-Strix-G10DK-G10DK-9619.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://soroushmaleki-ROG-Strix-G10DK-G10DK:37235/

SUMMARY
========

PARAMETERS
 * /ouster/os_driver/imu_frame: os_imu
 * /ouster/os_driver/imu_port: 0
 * /ouster/os_driver/lidar_frame: os_lidar
 * /ouster/os_driver/lidar_mode: 
 * /ouster/os_driver/lidar_port: 0
 * /ouster/os_driver/metadata: 
 * /ouster/os_driver/point_cloud_frame: 
 * /ouster/os_driver/point_type: original
 * /ouster/os_driver/proc_mask: IMG|PCL|IMU|SCAN
 * /ouster/os_driver/ptp_utc_tai_offset: -37.0
 * /ouster/os_driver/scan_ring: 0
 * /ouster/os_driver/sensor_frame: os_sensor
 * /ouster/os_driver/sensor_hostname: 169.254.218.69.161
 * /ouster/os_driver/tf_prefix: 
 * /ouster/os_driver/timestamp_mode: 
 * /ouster/os_driver/udp_dest: 
 * /ouster/os_driver/udp_profile_lidar: 
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    rviz (rviz/rviz)
  /ouster/
    os_driver (nodelet/nodelet)
    os_nodelet_mgr (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [9633]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bbf0b0f4-0e2c-11ef-9c13-7c10c952880f
process[rosout-1]: started with pid [9644]
started core service [/rosout]
process[ouster/os_nodelet_mgr-2]: started with pid [9647]
process[ouster/os_driver-3]: started with pid [9650]
process[rviz-4]: started with pid [9653]
[ INFO] [1715277077.140673394]: Loading nodelet /ouster/os_driver of type ouster_ros/OusterDriver to manager os_nodelet_mgr with the following remappings:
[ INFO] [1715277077.141730154]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] has not been advertised, waiting...
[ INFO] [1715277077.147409937]: Initializing nodelet with 12 worker threads.
[ INFO] [1715277077.162441423]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] is now available.
[ WARN] [1715277077.176938797]: lidar port set to zero, the client will assign a random port number!
[ WARN] [1715277077.176962722]: imu port set to zero, the client will assign a random port number!
[ INFO] [1715277077.176982209]: Will use automatic UDP destination
[FATAL] [1715277078.313069672]: Failed to load nodelet '/ouster/os_driver` of type `ouster_ros/OusterDriver` to manager `os_nodelet_mgr'
================================================================================REQUIRED process [ouster/os_nodelet_mgr-2] has died!
process has died [pid 9647, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=os_nodelet_mgr __log:=/home/soroushmaleki/.ros/log/bbf0b0f4-0e2c-11ef-9c13-7c10c952880f/ouster-os_nodelet_mgr-2.log].
log file: /home/soroushmaleki/.ros/log/bbf0b0f4-0e2c-11ef-9c13-7c10c952880f/ouster-os_nodelet_mgr-2*.log
Initiating shutdown!
================================================================================
[rviz-4] killing on exit
[ouster/os_driver-3] killing on exit
[ouster/os_nodelet_mgr-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`
```
I am sure that the sensor is working correctly and the serial number is not misspelled. I tested it with ouster studio and the point cloud is there and visualizable. I tried using two different ouster sensors namely ouster 0 and 1 and faced the same issue.

I would sincerely appreciate any suggestions.

**Platform (please complete the following information):**
- Ouster Sensor? OS-0 and OS-1
- Ouster Firmware Version? 2.5
- ROS version/distro? Melodic
- Operating System? Ubuntu 18
- Machine Architecture? x64
- git commit hash (if not the latest).

Samahu, 2024-05-10:
@SoroushMalekiXP we no longer officially support ROS melodic, please refer to the [comment](https://github.com/ouster-lidar/ouster-ros/issues/273#issuecomment-1847975235) for explanation.

In any case, you can still address the issue by implementing the following [fix](https://github.com/ouster-lidar/ouster_example/pull/571) submitted by one of our users, unfortunately, that fix wasn't merged :( so you would have to implement the fix in your local clone.

closed 2024-05-10 by Samahu

Samahu, 2024-05-10:
@SoroushMalekiXP I went ahead and merged the proposed fix, but to have this fully resolved i will need the ros-driver too.

SoroushMalekiXP, 2024-05-10:
Hey Ussama, @Samahu 

Thanks for your response. I implemented the fix and it resolved the initial compilation issue. However, I am still getting the exact same error when launching the driver through the driver.launch file:

```
... logging to /home/soroushmaleki/.ros/log/4b489d76-0edb-11ef-ac0e-7c10c952880f/roslaunch-soroushmaleki-ROG-Strix-G10DK-G10DK-7765.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://soroushmaleki-ROG-Strix-G10DK-G10DK:37681/

SUMMARY
========

PARAMETERS
 * /ouster/os_driver/imu_frame: os_imu
 * /ouster/os_driver/imu_port: 7503
 * /ouster/os_driver/lidar_frame: os_lidar
 * /ouster/os_driver/lidar_mode: 
 * /ouster/os_driver/lidar_port: 7502
 * /ouster/os_driver/metadata: 
 * /ouster/os_driver/point_cloud_frame: 
 * /ouster/os_driver/point_type: original
 * /ouster/os_driver/proc_mask: IMG|PCL|IMU|SCAN
 * /ouster/os_driver/ptp_utc_tai_offset: -37.0
 * /ouster/os_driver/scan_ring: 0
 * /ouster/os_driver/sensor_frame: os_sensor
 * /ouster/os_driver/sensor_hostname: os-122325001948
 * /ouster/os_driver/tf_prefix: 
 * /ouster/os_driver/timestamp_mode: 
 * /ouster/os_driver/udp_dest: 
 * /ouster/os_driver/udp_profile_lidar: 
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    rviz (rviz/rviz)
  /ouster/
    os_driver (nodelet/nodelet)
    os_nodelet_mgr (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [7779]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4b489d76-0edb-11ef-ac0e-7c10c952880f
process[rosout-1]: started with pid [7790]
started core service [/rosout]
process[ouster/os_nodelet_mgr-2]: started with pid [7793]
process[ouster/os_driver-3]: started with pid [7795]
process[rviz-4]: started with pid [7799]
[ INFO] [1715352050.060664983]: Loading nodelet /ouster/os_driver of type ouster_ros/OusterDriver to manager os_nodelet_mgr with the following remappings:
[ INFO] [1715352050.062214038]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] has not been advertised, waiting...
[ INFO] [1715352050.069965895]: Initializing nodelet with 12 worker threads.
[ INFO] [1715352050.082814201]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] is now available.
[ INFO] [1715352050.098199815]: Will use automatic UDP destination
[FATAL] [1715352051.174312812]: Failed to load nodelet '/ouster/os_driver` of type `ouster_ros/OusterDriver` to manager `os_nodelet_mgr'
**================================================================================REQUIRED process [ouster/os_nodelet_mgr-2] has died!
process has died [pid 7793, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=os_nodelet_mgr __log:=/home/soroushmaleki/.ros/log/4b489d76-0edb-11ef-ac0e-7c10c952880f/ouster-os_nodelet_mgr-2.log].
log file: /home/soroushmaleki/.ros/log/4b489d76-0edb-11ef-ac0e-7c10c952880f/ouster-os_nodelet_mgr-2*.log
Initiating shutdown!
================================================================================**
[rviz-4] killing on exit
[ouster/os_nodelet_mgr-2] killing on exit
[ouster/os_driver-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

The command I am usnig to launch:

`roslaunch ouster_ros driver.launch sensor_hostname:=os-122325001948
`

Where os-122325001948 is the sensor hostname. Any suggestions :((( ?

SoroushMalekiXP, 2024-05-10:
Thanks! adding the additinional .local to hostname worked! I really appreciate it :)))