ouster-lidar/ouster-ros #329: Failed to load nodelet '/ouster/os_driver` of type `ouster_ros/OusterDriver` to manager `os_nodelet_mgr'

zhangye1111, 2024-05-16:
I am using Ouster OS0-128 on Ubuntu 18  (ROS Melodic). When I run driver.launch, i receive the following error and the node fails to load:

SUMMARY
========

PARAMETERS
 * /ouster/os_driver/imu_frame: os_imu
 * /ouster/os_driver/imu_port: 0
 * /ouster/os_driver/lidar_frame: os_lidar
 * /ouster/os_driver/lidar_mode: 
 * /ouster/os_driver/lidar_port: 0
 * /ouster/os_driver/metadata: meta.json
 * /ouster/os_driver/point_cloud_frame: 
 * /ouster/os_driver/point_type: xyzir
 * /ouster/os_driver/proc_mask: IMG|PCL|IMU|SCAN
 * /ouster/os_driver/ptp_utc_tai_offset: -37.0
 * /ouster/os_driver/scan_ring: 0
 * /ouster/os_driver/sensor_frame: os_sensor
 * /ouster/os_driver/sensor_hostname: 192.168.100.105
 * /ouster/os_driver/tf_prefix: 
 * /ouster/os_driver/timestamp_mode: 
 * /ouster/os_driver/udp_dest: 
 * /ouster/os_driver/udp_profile_lidar: 
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    rviz (rviz/rviz)
  /ouster/
    os_driver (nodelet/nodelet)
    os_nodelet_mgr (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6919]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 32b1a5d2-135e-11ef-afe3-7404f1617ac4
process[rosout-1]: started with pid [6930]
started core service [/rosout]
process[ouster/os_nodelet_mgr-2]: started with pid [6934]
process[ouster/os_driver-3]: started with pid [6938]
process[rviz-4]: started with pid [6940]
[ INFO] [1715848077.376855219]: Loading nodelet /ouster/os_driver of type ouster_ros/OusterDriver to manager os_nodelet_mgr with the following remappings:
[ INFO] [1715848077.377677200]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] has not been advertised, waiting...
[ INFO] [1715848077.381853106]: Initializing nodelet with 12 worker threads.
[ INFO] [1715848077.398466015]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] is now available.
[ WARN] [1715848077.412820934]: lidar port set to zero, the client will assign a random port number!
[ WARN] [1715848077.412840483]: imu port set to zero, the client will assign a random port number!
[ INFO] [1715848077.412847673]: Will use automatic UDP destination
[ INFO] [1715848077.561746483]: Sensor 192.168.100.105 was configured successfully
[ INFO] [1715848077.561812646]: Starting sensor 192.168.100.105 initialization...
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/zwb/ouster_ros_ws/devel/lib/libouster_ros.so: undefined symbol: _ZN3fmt2v76detail10vformat_toIcEEvRNS1_6bufferIT_EENS0_17basic_string_viewIS4_EENS0_17basic_format_argsINS0_20basic_format_contextINS1_15buffer_appenderINS0_13type_identityIS4_E4typeEEESE_EEEENS1_10locale_refE
[FATAL] [1715848077.564863307]: Failed to load nodelet '/ouster/os_driver` of type `ouster_ros/OusterDriver` to manager `os_nodelet_mgr'
================================================================================REQUIRED process [ouster/os_nodelet_mgr-2] has died!
process has died [pid 6934, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=os_nodelet_mgr __log:=/home/zwb/.ros/log/32b1a5d2-135e-11ef-afe3-7404f1617ac4/ouster-os_nodelet_mgr-2.log].
log file: /home/zwb/.ros/log/32b1a5d2-135e-11ef-afe3-7404f1617ac4/ouster-os_nodelet_mgr-2*.log
Initiating shutdown!
================================================================================
[rviz-4] killing on exit
[ouster/os_driver-3] killing on exit
[ouster/os_nodelet_mgr-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

How do we fix this?

Samahu, 2024-05-16:
Seems to be related to this [issue](https://github.com/ouster-lidar/ouster-ros/issues/325#issuecomment-2103657513).

Samahu, 2024-10-17:
Although a workaround is possible and is outlined in the linked issue (and you are welcome to try it) but we no longer officially support **ROS Melodic**.

closed 2024-10-17 by Samahu