ouster-lidar/ouster-ros #337: [ROS2] Launching nodes when the lidar is not yet powered

Risbo6, 2024-06-06:
Hi,

I'm launching `driver.launch.py` as a service using systemd during the vehicle boot process. The issue is that the lidar powers up approximately 30 seconds later, causing the service to get stuck in a transitioning failed state. Restarting the service resolves the issue, but I would like this to be handled automatically.

Any ideas on how I can achieve this?

```
Mai 14 15:14:47 nvidia-desktop systemd[1]: Starting ROS2 Launch file for os-lidar...
Mai 14 15:17:14 nvidia-desktop systemd[1]: Started ROS2 Launch file for os-lidar.
Mai 14 15:17:19 nvidia-desktop bash[3880]: [INFO] [launch]: All log files can be found below /home/nvidia/.ros/log/2024-05-14-15-17-19-614101-nvidia-desktop-3880
Mai 14 15:17:19 nvidia-desktop bash[3880]: [INFO] [launch]: Default logging verbosity is set to INFO
Mai 14 15:17:20 nvidia-desktop bash[3880]: [INFO] [os_driver-1]: process started with pid [4308]
Mai 14 15:17:20 nvidia-desktop bash[3880]: [INFO] [ros2-2]: process started with pid [4310]
Mai 14 15:17:20 nvidia-desktop bash[3880]: [INFO] [ros2-3]: process started with pid [4312]
Mai 14 15:17:23 nvidia-desktop bash[3880]: [os_driver-1] [WARN] [1715692643.809827880] [ouster.os_driver]: lidar port set to zero, the client will assign a random port number!
Mai 14 15:17:23 nvidia-desktop bash[3880]: [os_driver-1] [WARN] [1715692643.810092849] [ouster.os_driver]: imu port set to zero, the client will assign a random port number!
Mai 14 15:17:23 nvidia-desktop bash[3880]: [os_driver-1] [INFO] [1715692643.810145920] [ouster.os_driver]: Will use automatic UDP destination
Mai 14 15:17:29 nvidia-desktop bash[3880]: [os_driver-1] [ERROR] [1715692649.049734280] [ouster.os_driver]: exception thrown while configuring the sensor, details: CurlClient::execute_get failed for the url: [http://os-122303000256.local/api/v1/system/firmware] with the error message: Couldn't resolve host name
Mai 14 15:17:29 nvidia-desktop bash[3880]: [os_driver-1] [WARN] [1715692649.049915402] []: Error occurred while doing error handling.
Mai 14 15:17:29 nvidia-desktop bash[3880]: [ros2-2] Transitioning failed
Mai 14 15:17:29 nvidia-desktop bash[3880]: [ros2-3] Unknown transition requested, available ones are:
Mai 14 15:17:30 nvidia-desktop bash[3880]: [ERROR] [ros2-3]: process has died [pid 4312, exit code 1, cmd 'bash -c sleep 1; $0 $@ /opt/ros/foxy/bin/ros2 lifecycle set /ouster/os_driver activate'].
Mai 14 15:17:30 nvidia-desktop bash[3880]: [INFO] [ros2-2]: process has finished cleanly [pid 4310]
```

**Platform (please complete the following information):**
OS-1
ROS 2 Foxy
Linux
ARM (jetson)

Samahu, 2024-06-10:
Definitely we want to handle this case .. but I am wondering if adding a delay to the node startup would help with you situation since you know how long it takes for the sensor to get powered up.

Samahu, 2024-08-09:
Hi @Risbo6 , as I noted before the driver doesn't handle re-connection if the sensor loses power or network is disconnected. However, one of our community members made a contribution a while back to this regard which I intend to look into soon and integrate in the ROS1 driver and then port it to ROS2, you may check it out here https://github.com/ouster-lidar/ouster-ros/pull/119 and see if covers your use case.

Stay tuned!

Samahu, 2024-08-31:
resolved by #362

PR #362 (2024-08-31): ROS-119: ouster-ros driver automatic reconnection [HUMBLE/IRON/JAZZY]
## Related Issues & PRs
- related #55
- resolves #119
- resolves #233
- related #332
- resolves #337

## Summary of Changes
- Implement sensor reconnection behavior
- Handle sensor reconfiguration from a cold start
- Add a flag to enable sensor reconnect (default disabled)
- properly handle config change outside of the ros driver
- only persist config startup config if persist_config was set
- only `auto_udp` if startup config didn't set an address
- added a parameter to set the dromant period between re-connection attempts
 [... cut]
closed 2024-08-31 by Samahu