ouster-lidar/ouster-ros #338: Failed to run /ouster/os_driver

Lin-Yi-Yang, 2024-06-11:
I use ros2 launch ouster_ros sensor.launch.xml sensor_hostname:=<sensor host name> to connect to lidar
The launch file works,but it seems the /ouster/os_driver finalized

[INFO] [launch]: All log files can be found below /home/juang/.ros/log/2024-06-11-09-20-14-531718-juang-Nuvo-7000-Series-4867
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [os_driver-1]: process started with pid [4869]
[INFO] [ros2-2]: process started with pid [4871]
[INFO] [ros2-3]: process started with pid [4874]
[INFO] [rviz2-4]: process started with pid [4878]
[os_driver-1] [WARN] [1718068816.697914279] [ouster.os_driver]: lidar port set to zero, the client will assign a random port number!
[os_driver-1] [WARN] [1718068816.698113521] [ouster.os_driver]: imu port set to zero, the client will assign a random port number!
[os_driver-1] [INFO] [1718068816.698162382] [ouster.os_driver]: Will use automatic UDP destination
[os_driver-1] [ERROR] [1718068816.704690524] [ouster.os_driver]: exception thrown while configuring the sensor, details: SensorHttp:: create firmware version information unavailable or not fully supported version. Please upgrade your sensor to FW 2.0 or later.
[os_driver-1] [WARN] [1718068816.704816676] []: Error occurred while doing error handling.
[ros2-2] /opt/ros/galactic/bin/ros2:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
[ros2-2]   from pkg_resources import load_entry_point
[ros2-2] Transitioning failed
[INFO] [ros2-2]: process has finished cleanly [pid 4871]
[ros2-3] /opt/ros/galactic/bin/ros2:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
[ros2-3]   from pkg_resources import load_entry_point
[ros2-3] Unknown transition requested, available ones are:
[ERROR] [ros2-3]: process has died [pid 4874, exit code 1, cmd 'bash -c sleep 1; $0 $@ /opt/ros/galactic/bin/ros2 lifecycle set /ouster/os_driver activate'].
[rviz2-4] [INFO] [1718068820.153196275] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1718068820.153254068] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1718068820.169938417] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1718068820.340253667] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2-4]: process has finished cleanly [pid 4878]

I check    ros2 node list
it shows /ouster/os_driver lives but finalized

So it has nothing in /ouster/points

How can I solve this without running each node independently?

Samahu, 2024-06-12:
The driver is unable to communicate with the sensors, did you configure your wired connection to "local only" as per the sensor setup guide
```
exception thrown while configuring the sensor, details: SensorHttp:: create firmware version information unavailable or not fully supported version. Please upgrade your sensor to FW 2.0 or later
```

Samahu, 2024-08-19:
hoping that my hint helped resolve the issue, closing the issue due to inactivity!

closed 2024-08-19 by Samahu