ouster-lidar/ouster-ros #341: Having trouble launching the sensor's driver in docker and getting messages

SoroushMalekiXP, 2024-06-27:
**Describe the bug**

Hello,

I am having trouble getting ROS messages from our OS0 and OS1 LiDARS in Docker. The problem begins when I build the docker image from the Dockerfile in ROS2 branch using the following command:

`docker build -t ouster_ros -f Dockerfile .`

Then I run the container using the following command:

`docker run --rm -it ouster-ros sensor_hostname:=os-serial_id.local
`
In this case I get an error saying that "Rviz2 is not installed" and this is in spite of running everything in docker using the dockerfile provided. To remedy, I disable the vis argument in the launch file and rebuild the image. In this case, if I use the os-serial_id.local as the sensor hostname I get an error saying that it could not resolve hostname. However, if I enter the IPv4 of the sensor instead for the hostname, it does find the sensor and it says "transaction successful". However, when I attach to the container in a separate terminal and run:

`ros2 topic echo /ouster/points`

I get no messages, meaning that nothing is being published. Perhaps there is something wrong with the way I am using the docker container and I would highly appreciate any help you may be able to provide. 

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. Build the image from this [dockerfile](https://github.com/ouster-lidar/ouster-ros/blob/2a0ef50f1dbe3f512321e809d6ca87391ce38f6e/Dockerfile#L1)
2. Run the container.
3. see if you can get messages

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
- Ouster Sensor? OS-0, OS-1
- Ouster Firmware Version? 2.5
- ROS version/distro? humble
- Operating System? Linux
- Machine Architecture? x64

Samahu, 2024-06-28:
@SoroushMalekiXP  regarding the RVIZ issue you would need to either disable rviz from running by setting the viz option:
```bash
docker run --rm -it ouster-ros sensor_hostname:=os-serial_id.local viz:=false
```
alternatively you can edit the Dockerfile and add the package `ros-humble-rviz2` to the install, however, if you plan to run gui with docker you need to pass a display context to when spawning a docker container.

Trying to connect to an ouster sensor from a docker container using the hostname doesn't work. It is mainly a dns discovery problem that isn't supported by the supplied Dockerfile, consider using IPv4 address for this case.

> However, when I attach to the container in a separate terminal and run. I get no messages,

Can you post the log output from the main launch? The sensor may have connected but the ROS node probably didn't activate!

SoroushMalekiXP, 2024-07-02:
Hey,

Thanks for getting back to me. I will post the output of the launch log below:

`
[INFO] [launch]: All log files can be found below /var/lib/build/.ros/log/2024-07-02-13-52-12-160414-e446a9a9e9dd-1
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [os_driver-1]: process started with pid [36]
[INFO] [ros2-2]: process started with pid [38]
[INFO] [ros2-3]: process started with pid [40]
[os_driver-1] [WARN] [1719928333.558731382] [ouster.os_driver]: lidar port set to zero, the client will assign a random port number!
[os_driver-1] [WARN] [1719928333.558801614] [ouster.os_driver]: imu port set to zero, the client will assign a random port number!
[os_driver-1] [INFO] [1719928333.558810711] [ouster.os_driver]: Will use automatic UDP destination
[os_driver-1] [INFO] [1719928333.579584232] [ouster.os_driver]: Sensor 192.168.50.224 configured successfully
[os_driver-1] [INFO] [1719928333.579627323] [ouster.os_driver]: Starting sensor 192.168.50.224 initialization...
[os_driver-1] [2024-07-02 13:52:13.579] [ouster::sensor] [info] initializing sensor client: 192.168.50.224 expecting lidar port/imu port: 0/0 (0 means a random port will be chosen)
[os_driver-1] [2024-07-02 13:52:22.046] [ouster::sensor] [info] parsing non-legacy metadata format
[os_driver-1] [2024-07-02 13:52:22.046] [ouster::sensor] [warning] No lidar profile found. Using LEGACY lidar profile
[os_driver-1] [2024-07-02 13:52:22.046] [ouster::sensor] [warning] No imu profile found. Using LEGACY imu profile
[os_driver-1] [INFO] [1719928342.046680911] [ouster.os_driver]: No metadata file was specified, using: 192.168.50-metadata.json
[os_driver-1] [INFO] [1719928342.046890984] [ouster.os_driver]: Wrote sensor metadata to 192.168.50-metadata.json
[os_driver-1] [INFO] [1719928342.046916662] [ouster.os_driver]: ouster client version: 0.10.0+unknown-release
[os_driver-1] product: OS-1-16-A1, sn: 991951001157, firmware rev: v2.0.0
[os_driver-1] lidar mode: 1024x10, lidar udp profile: LEGACY
[os_driver-1] [INFO] [1719928342.049695238] [ouster.os_driver]: reset service created
[os_driver-1] [INFO] [1719928342.050199824] [ouster.os_driver]: get_metadata service created
[os_driver-1] [INFO] [1719928342.050547024] [ouster.os_driver]: get_config service created
[os_driver-1] [INFO] [1719928342.050785892] [ouster.os_driver]: set_config service created
[ros2-2] Transitioning successful
[ros2-3] Transitioning successful
[INFO] [ros2-2]: process has finished cleanly [pid 38]
[INFO] [ros2-3]: process has finished cleanly [pid 40]
`

It is worth mentioning that the list of available topics after launching is:

`
/ouster/imu
/ouster/metadata
/ouster/nearir_image
/ouster/os_driver/transition_event
/ouster/points
/ouster/range_image
/ouster/reflec_image
/ouster/scan
/ouster/signal_image
/parameter_events
/rosout
/tf_static
`
Is there anything missing here? I remember having an ouster/points2 topic in the ros1 driver. If so, would this be a hardware problem or a software problem?

Thank you for your help and attention.

Samahu, 2024-07-02:
The `ouster/points2` would only be published if you are using the DUAL mode in the list of udp profiles.

> Finally, I called the get_config service inside the container to see if the services work and I got the following error:

Did you source the `install/setup.bash` of the driver?

SoroushMalekiXP, 2024-07-02:
Yes, I have sourced the install/setup.bash when I attach to it. It also sources it by default in the entry point of the dockerfile. I can call the get status service and see that the lidar is "active". 
How can I set the mode to DUAL? Perhaps that might do something. But I'm feeling a bit desperate right now :)
Could this be an issue with the LiDAR itself? Or the firmware version?

Samahu, 2024-07-02:
The reason I asked if you source the environment because I don't understand this error

> ModuleNotFoundError: No module named 'ouster_sensor_msgs

You should have the **ouster_sensor_msgs** package in your system, can you verify it exists in the source tree that you are building from.

SoroushMalekiXP, 2024-07-02:
I am sorry Usama, I think the issue with the service is resolved by simply reattaching to the container and resourcing. However, I still cannot get any points. This is the output of the get config service:

```
ouster_sensor_msgs.srv.GetConfig_Response(config='{\n    "azimuth_window": \n    [\n        0,\n        360000\n    ],\n    "lidar_mode": "1024x10",\n    "multipurpose_io_mode": "OFF",\n    "nmea_baud_rate": "BAUD_9600",\n    "nmea_ignore_valid_char": 0,\n    "nmea_in_polarity": "ACTIVE_HIGH",\n    "nmea_leap_seconds": 0,\n    "operating_mode": "NORMAL",\n    "phase_lock_enable": false,\n    "phase_lock_offset": 0,\n    "sync_pulse_in_polarity": "ACTIVE_HIGH",\n    "sync_pulse_out_angle": 360,\n    "sync_pulse_out_frequency": 1,\n    "sync_pulse_out_polarity": "ACTIVE_HIGH",\n    "sync_pulse_out_pulse_width": 10,\n    "timestamp_mode": "TIME_FROM_INTERNAL_OSC",\n    "udp_dest": "192.168.50.200",\n    "udp_port_imu": 50042,\n    "udp_port_lidar": 39025\n}
```

SoroushMalekiXP, 2024-07-04:
Hey Usuama,

While I was trying different things to get this to work, I installed the driver on the base os without docker and ran it using ros2 launch command. I got a different error saying that the node is not found!!!

```
[INFO] [launch]: All log files can be found below /home/soroushmaleki/.ros/log/2024-07-04-10-22-57-293508-soroushmaleki-ROG-Strix-G10DK-G10DK-304886
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [os_driver-1]: process started with pid [304888]
[INFO] [ros2-2]: process started with pid [304890]
[INFO] [ros2-3]: process started with pid [304892]
[INFO] [rviz2-4]: process started with pid [304896]
[ros2-2] Node not found
[ERROR] [ros2-2]: process has died [pid 304890, exit code 1, cmd 'bash -c sleep 0; $0 $@ /opt/ros/foxy/bin/ros2 lifecycle set /ouster/os_driver configure'].
[ros2-3] Node not found
[ERROR] [ros2-3]: process has died [pid 304892, exit code 1, cmd 'bash -c sleep 1; $0 $@ /opt/ros/foxy/bin/ros2 lifecycle set /ouster/os_driver activate'].
[rviz2-4] [INFO] [1720102982.991356547] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1720102982.991473096] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1720102983.084273972] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1720102983.173730310] [rviz2]: Stereo is NOT SUPPORTED

```

I would highly appreciate any insight you may be able to provide on this...

Samahu, 2024-07-08:
Which launch file are you using to start the driver up?

Samahu, 2024-07-08:
also regarding launch using docker .. I have verified it works correctly on humble and foxy.

SoroushMaleki, 2024-07-11:
I see. Could you please write a step by step guide on how you attached to the container and verified that you're getting the /ouster/points topic? when I run the docker container and attach to it and type "ros2 topic echo /ouster/points" it waits for a few seconds and prints nothing.

Samahu, 2024-07-11:
Build your docker image, make sure to sure either to copy or attach your driver_params.yaml to the instantiated container
```bash
docker run --rm -it ros-docker-image /bin/bash
```
Inside the container terminal 
```bash
source /opt/ros/$ROS_DISTRO/setup.bash
source install/setup.bash # where ouster-ros install located
ros2 launch ouster_ros driver.launch.py viz:=false
```

In another terminal connect to the container
````bash
docker exec --it <container> /bin/bash
```
Inside the container terminal 
```bash
source /opt/ros/$ROS_DISTRO/setup.bash
source install/setup.bash # where ouster-ros install located
ros2 topic echo /ouster/points
```

closed 2024-07-11 by Samahu