ouster-lidar/ouster-ros #342: camera_info for 2d images

bracoe, 2024-06-28:
Hello,

I want to use either the OSDome or OS0 together with the standard ROS2 image_pipeline packages. Especially [depth_image_proc](https://github.com/ros-perception/image_pipeline/tree/rolling/depth_image_proc) so I can give depth to an RGB image.  But without a camera_info it doesn't really work... How would one go about doing this?

**Describe the solution you'd like**
The release of a camera_info topic with the correct values for using [depth_image_proc](https://github.com/ros-perception/image_pipeline/tree/rolling/depth_image_proc).

**Describe alternatives you've considered**
Calibrating it with [camera_calibration](https://github.com/ros-perception/image_pipeline/tree/rolling/camera_calibration) for a specific section in the range/reflectivity 2D image. But I have no idea if that will work considering the 360 degrees.

- Ouster Sensor? OSDome/OS0 (preferably all sensors)
- Ouster Firmware Version? Doesn't matter as long as i can update!
- ROS version/distro? iron and later releases. 
- Operating System? Ubuntu
- Machine Architecture? amd64 or arm64

sepsamavi, 2025-02-12:
Any updates on this?

bracoe, 2025-02-13:
Still waiting for this as well

Samahu, 2025-02-19:
Sorry for the delay in response, I created the PR #441 for tracking but I haven't researched the parameters in details (yet).

nathantsoi, 2026-04-19:
+1 for publishing the camera_info topic