ouster-lidar/ouster-ros #343: Bond Broken playing bag files Orin Nano Ubuntu 20.04

Rotoslider, 2024-07-08:
I have played these bags bag on previous Jetsons including standard Nano's to a AGX with no issue. New build on Orin Nano. Ubuntu 20.04. Latest Ouster SDK. I get Bond broken when trying to playback any bags. All running local on Orin. 

It plays a couple seconds in Rviz then fails and closes.
`$ roslaunch ouster_ros replay.launch viz:=true metadata:=/home/lidar/pointclouds/mato/mato.json bag_file:=/home/lidar/pointclouds/mato/mato.bag  
... logging to /home/lidar/.ros/log/35818a5c-3b22-11ef-a01f-1cce518e7269/roslaunch-ouster-12744.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ouster:38527/

SUMMARY
========

PARAMETERS
 * /ouster/os_cloud_node/dynamic_transforms_broadcast: False
 * /ouster/os_cloud_node/dynamic_transforms_broadcast_rate: 1.0
 * /ouster/os_cloud_node/imu_frame: os_imu
 * /ouster/os_cloud_node/lidar_frame: os_lidar
 * /ouster/os_cloud_node/point_cloud_frame: 
 * /ouster/os_cloud_node/point_type: original
 * /ouster/os_cloud_node/proc_mask: IMG|PCL|IMU|SCAN
 * /ouster/os_cloud_node/ptp_utc_tai_offset: -37.0
 * /ouster/os_cloud_node/scan_ring: 0
 * /ouster/os_cloud_node/sensor_frame: os_sensor
 * /ouster/os_cloud_node/tf_prefix: 
 * /ouster/os_cloud_node/timestamp_mode: 
 * /ouster/os_node/metadata: /home/lidar/point...
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    rosbag_play_recording (rosbag/play)
    rviz (rviz/rviz)
  /ouster/
    img_node (nodelet/nodelet)
    os_cloud_node (nodelet/nodelet)
    os_node (nodelet/nodelet)
    os_nodelet_mgr (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [12753]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 35818a5c-3b22-11ef-a01f-1cce518e7269
process[rosout-1]: started with pid [12763]
started core service [/rosout]
process[ouster/os_nodelet_mgr-2]: started with pid [12770]
process[ouster/os_node-3]: started with pid [12772]
process[ouster/os_cloud_node-4]: started with pid [12774]
process[ouster/img_node-5]: started with pid [12775]
process[rviz-6]: started with pid [12776]
[ INFO] [1720220368.707495204]: Loading nodelet /ouster/os_cloud_node of type ouster_ros/OusterCloud to manager os_nodelet_mgr with the following remappings:
[ INFO] [1720220368.712615994]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] has not been advertised, waiting...
process[rosbag_play_recording-7]: started with pid [12783]
[ INFO] [1720220368.719303635]: Loading nodelet /ouster/img_node of type ouster_ros/OusterImage to manager os_nodelet_mgr with the following remappings:
[ INFO] [1720220368.721076187]: Initializing nodelet with 6 worker threads.
[ INFO] [1720220368.723434705]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] has not been advertised, waiting...
[ INFO] [1720220368.739056183]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] is now available.
[ INFO] [1720220368.749172959]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] is now available.
[ INFO] [1720220368.770743533]: OusterCloud: nodelet created!
[ INFO] [1720220368.780046370]: OusterImage: node initialized!
[ INFO] [1720220371.700632838]: Loading nodelet /ouster/os_node of type ouster_ros/OusterReplay to manager os_nodelet_mgr with the following remappings:
[2024-07-05 15:59:31.717] [ouster::sensor] [info] parsing legacy metadata format
[ INFO] [1720220371.718195096]: ouster client version: 0.10.0+2898060-release
product: OS-1-32-U3, sn: 122223000838, firmware rev: v2.5.1
lidar mode: 512x10, lidar udp profile: RNG19_RFL8_SIG16_NIR16
[ INFO] [1720220371.718525311]: OusterCloud: retrieved new sensor metadata!
[2024-07-05 15:59:31.718] [ouster::sensor] [info] parsing legacy metadata format
[ INFO] [1720220371.719235632]: get_metadata service created
[ INFO] [1720220371.719372083]: Running in replay mode
[ INFO] [1720220371.719643385]: OusterImage: retrieved new sensor metadata!
[2024-07-05 15:59:31.719] [ouster::sensor] [info] parsing legacy metadata format
[ WARN] [1720220371.727407306]: point_cloud_frame is set to sensor_frame. If you need  to reproject or raytrace the points then use the lidar_frame instead. Refer to: https://github.com/ouster-lidar/ouster-ros/issues/33 for more details.
[ INFO] [1720220371.731597642]: OusterCloud: using static transforms broadcast
[ INFO] [1720220371.752185953]: Opening /home/lidar/pointclouds/mato/mato.bag

Waiting 0.2 seconds after advertising topics... done.

Hit space to toggle paused, or 's' to step.
[ INFO] [1720220374.021904339]: rviz version 1.14.251.821192 / 306.311608               261.46 
[ INFO] [1720220374.022026262]: compiled against Qt version 5.12.8
[ INFO] [1720220374.022049399]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1720220374.035495081]: Forcing OpenGl version 0.414 / 306.311608               
[ INFO] [1720220374.716738873]: Stereo is NOT SUPPORTED15379 / 306.311608               
[ INFO] [1720220374.716893661]: OpenGL device: NVIDIA Tegra Orin (nvgpu)/integrated
[ INFO] [1720220374.716955326]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1720220381.824058517]: Bond broken, exiting9.621345 / 306.311608               
================================================================================REQUIRED process [ouster/os_cloud_node-4] has died!
process has finished cleanly
log file: /home/lidar/.ros/log/35818a5c-3b22-11ef-a01f-1cce518e7269/ouster-os_cloud_node-4*.log
Initiating shutdown!
================================================================================
[rosbag_play_recording-7] killing on exit Duration: 9.963876 / 306.311608               
[rviz-6] killing on exit
[ouster/os_cloud_node-4] killing on exit
[ouster/img_node-5] killing on exit1647   Duration: 9.973529 / 306.311608               
[ouster/os_node-3] killing on exit
[ouster/os_nodelet_mgr-2] killing on exit
[ INFO] [1720220382.176628945]: Unloading nodelet /ouster/img_node from manager os_nodelet_mgr
[ INFO] [1720220382.176752788]: Unloading nodelet /ouster/os_node from manager os_nodelet_mgr
 [RUNNING]  Bag Time: 1692445120.862160   Duration: 10.224042 / 306.311608               
[ouster/os_nodelet_mgr-2] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

Samahu, 2024-07-08:
First time I have seen this issue on non-live setup. This indicates that there are some synchronization issues between the different ros nodes on the system. Usually this is an issue when running on ROS on multiple machines but could probably happen if the machine is running slow.

In any case, the driver implements a no_bond flag in `common.launch.xml` and is used in `sensor.launch.xml` and some other files but not in `replay.launch.xml`. You can try to add this flag and feed it to the include `common.launch.xml` in `replay.launch.xml` and check if this helps with your case.

Samahu, 2024-07-19:
Hoping that my answer helped. Closing the issue due to inactivity.

closed 2024-07-19 by Samahu