ouster-lidar/ouster-ros #345: Ros2 humble node not found error

curiousityy, 2024-07-17:
Hi Team,

I am currently testing the ouster driver with ROS2 on ARM64 with docker. When I run the driver within the docker image(built using the Dockerfile in the repo), I get the below error.

[INFO] [launch]: All log files can be found below /var/lib/build/.ros/log/2024-07-17-15-04-34-154041-ubuntu-74
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [os_driver-1]: process started with pid [75]
[INFO] [bash-2]: process started with pid [77]
[INFO] [bash-3]: process started with pid [79]
[bash-2] Node not found
[ERROR] [bash-2]: process has died [pid 77, exit code 1, cmd 'bash -c sleep 0; $0 $@ /opt/ros/humble/bin/ros2 lifecycle set /ouster/os_driver configure'].
[bash-3] Node not found
[ERROR] [bash-3]: process has died [pid 79, exit code 1, cmd 'bash -c sleep 1; $0 $@ /opt/ros/humble/bin/ros2 lifecycle set /ouster/os_driver activate'].

I am able to ping the ouster within the container and I also tested with ROS1 and that works too. The problem I face is only for ROS2. Can you help me with the same

Samahu, 2024-07-19:
Either increase the sleep period in the launch file `cmd 'bash -c sleep 0`  or it would be better to switch to using `driver.launch.py`. The later uses events so it does't need to set sleep period.

closed 2024-07-19 by Samahu

curiousityy, 2024-07-26:
@Samahu, thanks for the update. I switched to the driver.launch.py and the results are more consistent in terms of launching. However, the publish artes are very low. I chose a 1024x10 configuration for the pointcloud, but the publish rate is hovering around 6-7Hz

Samahu, 2024-07-26:
The low frame rate might be related to this: https://github.com/ouster-lidar/ouster-ros/issues/346, you may try some of the options that I have recommend.

James-R-Han, 2024-07-26:
I have also found a similar issue that with the run command:
`ros2 launch ouster_ros sensor.launch.xml sensor_hostname:=192.168.131.18 use_system_default_qos:=true sensor_qos_profile:=reliable`

I'll get the error:
`[INFO] [launch]: All log files can be found below /home/jha/.ros/log/2024-07-26-13-00-32-727572-jha-ThinkPad-T15g-Gen-2i-392187
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [os_driver-1]: process started with pid [392195]
[INFO] [ros2-2]: process started with pid [392197]
[INFO] [ros2-3]: process started with pid [392200]
[ros2-2] Node not found
[ERROR] [ros2-2]: process has died [pid 392197, exit code 1, cmd 'bash -c sleep 0; $0 $@ /opt/ros/galactic/bin/ros2 lifecycle set /ouster/os_driver configure'].
[ros2-3] Unknown transition requested, available ones are:
[ros2-3] - configure [1]
[ros2-3] - shutdown [5]
[ERROR] [ros2-3]: process has died [pid 392200, exit code 1, cmd 'bash -c sleep 1; $0 $@ /opt/ros/galactic/bin/ros2 lifecycle set /ouster/os_driver activate'].
`

I need to run it again for the driver to work. Once it's running it's okay, but I thought I'd point out this issue.

curiousityy, 2024-08-13:
Hi @Samahu @James-R-Han,

Thanks for the suggestions. Unfortunately, the qos and rmw_implementation still don't solve the problem. The topics drop to 1-2Hz after 30-40s after starting off with the correct publish rate.

curiousityy, 2024-08-19:
Hi @Samahu,

Sorry for not making the point clear. Despite switching to CycloneDDS, the publish rate issue has not been solved. The publish rate is still less than 5Hz even though the LiDAR configuration is 1024x10.

Samahu, 2024-08-19:
Hi @curiousityy, and sorry for the delayed response. I was able to re-produce the problem on lower end compute boards and looking into it. I do have plans to improve the performance a bit more as captured in the TODOs list of this fix https://github.com/ouster-lidar/ouster-ros/pull/319 and also skip the middle step of having to convert to LidarScan before getting a PointCloud object, optimize the cartesian function plus more many other tweaks and improvements which should help the situation for you and others. You can keep track of this ticket https://github.com/ouster-lidar/ouster-ros/issues/352 for an update