ouster-lidar/ouster-ros #346: OS0 /ouster/points low frequency with rmw_fastrtps_cpp

James-R-Han, 2024-07-25:
Hello!

I currently have an OS0-128 Rev7 which I run with configuration 1024x10 and command: `ros2 launch ouster_ros sensor.launch.xml sensor_hostname:=192.168.131.18 use_system_default_qos:=true timestamp_mode:=TIME_FROM_ROS_TIME` I am using the Humble driver.

Sometimes it is quite fast and I see /ouster/points reach ~7-8 Hz but other times it reaches around 1Hz (see image below).

#278 uses CycloneDDS but some other nodes I'm running use rmw_fastrtps_cpp so I would like to avoid switching DDS if possible.
#63 suggests using sensor.independent.launch.xml which does help a bit but does not reach 10Hz.

Are there any other potential solutions I could try when using rmw_fastrtps_cpp? Also, I'm curious why the fluctuation can be so high (even though I test when the only thing I'm running is the sensor driver).

Thank you in advance!

[image]

Samahu, 2024-07-25:
Since the problem is mainly specific to fastdds it is safe to assume it DDS issue rather than the driver implementation in itself. But I understand that the point cloud the driver generates is large that open dds implementation doesn't currently work well under it. So consider the following:
* Use lower lidar mode
* Use the low data udp profile
* Consider customizing the generated point cloud and only use the fields that you need from the sensor
  -  refer to https://github.com/ouster-lidar/ouster-ros/discussions/260

Obviously, certain aspects can be improved from our side and we are constantly improving our code. But these are some of the steps that you can apply right now to mitigate the issue.

Hope this helps!

Samahu, 2024-07-26:
On my laptop and other computers I tested with I can hit ~20Hz, for the most part however I recommend people to use cyclonedds. Again, if you want to stick with fastdds you can try the options I enumerated above and see if they can help boost the frame rate. We have many improvements planned that may or may not help with fastdds.

James-R-Han, 2024-07-26:
Hey Samahu, I took your advice and switched to cyclonedds and I'm able to hit 10 Hz consistently now. Thank you!

closed 2024-07-26 by James-R-Han