ouster-lidar/ouster-ros #347: CycloneDDS results in 0s in point cloud

James-R-Han, 2024-07-26:
Hello!

After switching to CycloneDDS, although I can now hit 10Hz on /ouster/points, I find that I receive thousands of 0s in my point cloud. I have verified that with using FastRTPS, I don't get the same issue. I decode the message in the following way:

`self.qos_profile = QoSProfile(reliability=ReliabilityPolicy.RELIABLE, history=HistoryPolicy.KEEP_LAST, depth=10)`
`self.points_subscription = self.create_subscription(PointCloud2, '/ouster/points' , self.points_listener_callback, self.qos_profile)`
`import ros2_numpy as rnp`
`pc_array = rnp.numpify(msg)['xyz']`
`pc_array = pc_array.reshape(msg.height, msg.width, 3)`

Also, some other notes:

- When using CycloneDDS, I don't get any nan values in the point cloud
- When using FastRTPS, I get nans, but I don't get all 0 values.

Would anyone happen to have insight into this with the driver?

On a separate note, rather than construct my own range image in pc_array, how could we decode ouster-ros's range sensor_msgs.msg.Image into the same format? (I've been having trouble with the 19-bit representation to Euclidean conversion)

Sorry for the long question; I'd appreciate any help on either matter. Thank you [Samahu probably :)] in advance!

Samahu, 2024-07-26:
This is the first time that someone reported such an issue. I am running cyclonedds on my ubuntu machine and getting point clouds as usual. Which platform/ros-distro/architecture do you use?

> On a separate note, rather than construct my own range image in pc_array, how could we decode ouster-ros's range sensor_msgs.msg.Image into the same format? (I've been having trouble with the 19-bit representation to Euclidean conversion)

Not sure I fully got what you are trying to do but if you are using the /image_range topic this unfortunately truncates the image to 16 bit which isn't idea and causes issues. In any case, make sure to check this script in case you haven't already https://github.com/ouster-lidar/ouster-ros-extras/blob/ros2/scripts/ros2_range_2_points_xyz.py

James-R-Han, 2024-07-29:
Ubuntu 22.04, Humble, x86_64
Once I remove all the points with channels all equal to 0, the data is a lot more what I expect. I tried searching it up but wasn't able to find anything - potentially Nans in ros2_numpy + FastDDS is the same as 0s in ros2_numpy + CycloneDDS?

Ah thank you Samahu! I didn't know there was a helper function to already convert the range image to xyz for me.

Samahu, 2024-07-29:
Yes, the driver sets range value that had no returns to NANs which may not work with certain methods and thus you have to filter them out. But i wouldn't expect this to differ from one DDS to another. For example, I don't have this issue on my end when viewing the generate pointclouds on RVIZ. Perhaps one of your python numpy script have issues with NAN values.

James-R-Han, 2024-07-29:
I didn't change any of the python code when switching between FastDDS and CycloneDDS, so I don't think that's the issue.

Since the point cloud data looks acceptable after filtering out these 0 points (<4% of the point cloud), I don't think it's a huge issue. I thought I may have just missed some configuration when switching to CycloneDDS.

Thanks for taking a look into this Samahu!

closed 2024-07-29 by James-R-Han