ouster-lidar/ouster-ros #348: OS1 ROS not reading new messages after 1.5-2 hours of running

Jeli04, 2024-07-30:
I was running some experiments with OS1 Lidar, but noticed that after 1.5-2 hours of running ROS messages related with the lidar weren't being received anymore and the rviz visualizer would stop updating. No error messages appear when this happens. I've repeated this three times already and I'm not sure if this is a faulty hardware issue or something software related. 

Ouster Sensor == OS1
firmware version == 2.1.2
ROS version == noetic 
OS == Ubuntu 20.04

Samahu, 2024-07-31:
When you get in this state, do you see any errors list on the sensor diagnostic page?

To isolate the issue when this happen again, can you use **Wireshare** (or any packet sniffing software) to see if the packets are still coming from the sensor.

Jeli04, 2024-07-31:
I'll reply again with what I get from that after I recreate this issue again.

Jeli04, 2024-07-31:
when I try to nc os-XXXXXXXXXXX 7501 I keep getting an error "Temporary failure in name resolution"

Samahu, 2024-07-31:
Try `nc os-XXXXXXXXXXX.local`

Jeli04, 2024-07-31:
That worked thanks still waiting on it too stop receiving

Jeli04, 2024-07-31:
> Try `nc os-XXXXXXXXXXX.local 7501`

So it seems like I'm unable to query to the LiDAR and there aren't anymore packets coming in.

Samahu, 2024-08-01:
Oh ok, good to know so you may want to check the diagnostics on your sensor page (if still accessible) and look for any errors.

Jeli04, 2024-08-01:
> Oh ok, good to know so you may want to check the diagnostics tab on your sensor page (if still accessible) and look for any errors.

The sensor homepage doesn't open when this happens

Samahu, 2024-08-01:
Ok, in this case you may want to reach [Ouster support team](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86). Closing this ticket as this is a sensor issue.

closed 2024-08-01 by Samahu