ouster-lidar/ouster-ros #350: 't' timestamp field content is not plausible

lukasj-imar, 2024-08-06:
**Describe the bug**
I'm switching from community driver to this driver. In my application I use the content of the `t` data field for pointcloud deskewing. The value of this field by the community driver contents the relative timeshift of each point since start of frame in nanoseconds. So for a OS1-128 in 1024x20 operation mode, the timestamp of the first and each 1024th point is zero. The ofifcial driver outputs the following timestamp values for the corresponding clouds:

`point[0] = 49416704`
`point[12] = 0`
`point[1024] = 49806592`

I don't have any idea of the meaning of these timestamps. Please let me know if you need further information.

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. Configure with 'native' or 'original pointcloud profile'
2. compile the project workspace
3. source the project workspace
4. ros launch ouster_ros sensor or replay
5. open another terminal and the project workspace
6. observe the issue ...

**Platform (please complete the following information):**
- OS1-128 rev. 7
- Firmware v3.1.0
- ROS2 humble
- Ubuntu 22.04
- arch x86
- Release ros2-v0.12.0

Samahu, 2024-08-08:
Hi @lukasj-imar,

I want to look further into this but my gut feeling this happens when the lidar packet stream have some missing packets which means there is no proper time for these points. You can tell if this is the case by examining any of the XYZ values for any NANs.
Another occasion of when this could happen is if you are using PTP clock and a sync event causes the shift to occur mid of the lidarscan. This would definitely result in some of the timestamps being zeros. What clock do you use?

Samahu, 2024-10-01:
Hi @lukasj-imar, thanks for the feedback. I will look into this problem.

Samahu, 2024-10-08:
Hi @lukasj-imar, I am taking a stab at this problem and have few questions to help me better understand the situation:

For the data sample you provide:
```
point[0] = 49416704
point[12] = 0
point[1024] = 49806592
```
Are these timestamps of the same frame? the issue I see is that point[1024] is at 49806592 (nanoseconds) while point[0] = 49416704 (nanoseconds) and the difference is only 389888 nanoseconds ~ 390 microseconds .. meanwhile for a regular 1024x10 the difference should be 100000000 nanoseconds ~ 100 milliseconds.. Am reading the data correctly? If so the problem might be in the connected clock. Do you get something similar when using the sensor alone without external clock source?

lukasj-imar, 2024-10-08:
Hi @Samahu,

yes the timestamps are from the same frame, but the sensor is in 1024x20 mode, so `point[1024]` is round about in the right region of 50ms.

To see the influence of the external clock, I will do the same with the internal clock only and provide these results here asap.

Samahu, 2024-10-09:
@lukasj-imar Just for your information I do see some discrepancy in the `t` values even when using the internal clock, I am still digging to find out the source of it.

closed 2024-10-11 by Samahu

Samahu, 2024-10-11:
@lukasj-imar I have merged a fix for the problem, the timestamp values were being destaggered when they shouldn't be, which made the points and their timestamps inconsistent. The values should be aligned correctly whether you choose to consume a destaggered point cloud (the default) or a staggered one.

Note if you get zero values this is usually due to no returns in that spot in which the xyz values are NaNs and you should skip these values in your pipeline or use a non-organized point cloud which would skip the NaNs from the generated point cloud (if it makes sense for your pipeline).

Please let me know if you still observe an issue related to this topic.

ctampier, 2025-02-04:
Hello,

I'm facing problems with point cloud de-skewing. They started **after** this PR was merged. Before I didn't have the issue.
The previous de-staggering of timestamps (point_cloud_compose.h:140) works correctly in my case:
```
auto ts = timestamp[v_shift];
ts = ts > scan_ts ? ts - scan_ts : 0UL;
```
I also makes a sense to me that points and timestamps are shifted equally. For any row in a de-staggered cloud, the "t" point field should be minimum at the "pixel offset position" for the given row, since that was the first fired laser. That's the result I was getting with the previous code.

I am using the ouster driver for ROS1 and with  "timestamp_mode" = "TIME_FROM_ROS_TIME" tho.

Samahu, 2025-02-04:
Hi @ctampier, have you seen [this comment](https://github.com/ouster-lidar/ouster-ros/issues/423#issuecomment-2611362018). I am referring to this one specifically because it shows you the structure of timestamps in a staggered vs destaggered PointCloud. also it advised against the use of `TIME_FROM_ROS_TIME` as it does have an issue which is yet to be fixed.

ctampier, 2025-02-05:
Hi @Samahu,
Thanks for the quick response and the provided info. Like the user in the comment you linked, I too think the "t" point field in a staggered and structured point cloud should increase monotonically for each row. And it should be shifted, same as the other data fields, when destaggered.
I fail to understand any other scenario, since shifting one but not the other results in incorrect time data in all point cloud points.
In a point cloud deskewing operation, I would take the point cloud header stamp + point "t" data to obtain the absolute time at which a point was measured. Doing this, I was previously able to correctly deskew the point cloud. Now, I'm getting incorrect result, which are especially noticeable at the beginning of each ring (of a destaggered cloud), up until the row pixel offset.
Whether this is an isolated issue when using TIME_FROM_ROS_TIME, I cannot tell. For now I have to stick to this mode, since it's the only one that generates compatible stamps to my other sensor data.

Samahu, 2025-02-05:
> Thanks for the quick response and the provided info. Like the user in the comment you linked, I too think the "t" point field in a staggered and structured point cloud should increase monotonically for each row. And it should be shifted, same as the other data fields, when destaggered.

I am not sure if you read the whole thread but the user in the linked thread actually agrees that the timestamps of a staggered PointCloud should use the point capture time.

> I fail to understand any other scenario, since shifting one but not the other results in incorrect time data in all point cloud points.
In a point cloud deskewing operation, I would take the point cloud header stamp + point "t" data to obtain the absolute time at which a point was measured. Doing this, I was previously able to correctly deskew the point cloud. Now, I'm getting incorrect result, which are especially noticeable at the beginning of each ring of a destaggered cloud, up until the row pixel offset (that's where the jump or difference due to unshifted timestamps is greatest).

I am a bit lost, are you using a destaggered point cloud or a staggered point cloud? When using a destaggered point cloud the timestamps would be increasing monotonically.

> Whether this is an isolated issue when using TIME_FROM_ROS_TIME, I cannot tell. For now I have to stick to this mode, since it's the only one that generates stamps compatible with my other sensor data.

Try observing the PointCloud timestamps from other modes first. The issue with using the `TIME_FROM_ROS_TIME` is described here https://github.com/ouster-lidar/ouster-ros/issues/407#issuecomment-2504404475. The issue is with regard to `TIME_FROM_ROS_TIME`  it only marks the PointCloud message time but leaves the frame timestamp intact, and thus when you take the relative timestamp (as it is the case with the other modes) this can lead to a bunch of zeros generates. So this needs to be fixed. I will try to open a fix soon.

mjseppa, 2025-12-15:
Noted this in the actual PR, but the "t" field now has incorrect data:  
https://github.com/ouster-lidar/ouster-ros/pull/387#issuecomment-3654915749

PR #386 (2024-10-11): ROS-350[NOETIC]: 't' timestamp field content is not plausible
## Related Issues & PRs
- resolves #350

## Summary of Changes
- Don't apply offset to destaggered timestamps within the point cloud
- Make sure the the timestamps align correctly when destaggerd is option is OFF
- Misc: update bug/question templates

## Validation
* run the driver with default option (destaggered=ON, organized=ON)
  -  observe that the timestamps are consistent
* run the driver with default option (destaggered=OFF, organized=ON)
  - observe that the timestamps appear staggerd
PR #387 (2024-10-11): ROS-350[HUMBLE/IRON/JAZZY]: 't' timestamp field content is not plausible
## Related Issues & PRs
- resolves #350

## Summary of Changes
- Don't apply offset to destaggered timestamps within the point cloud
- Make sure the the timestamps align correctly when destaggerd is option is OFF

## Validation
* run the driver with default option (destaggered=ON, organized=ON)
  -  observe that the timestamps are consistent
* run the driver with default option (destaggered=OFF, organized=ON)
  - observe that the timestamps appear staggerd