ouster-lidar/ouster-ros #351: Bug with Ouster in ROS Noetic

yulan0215, 2024-08-06:
**Describe the bug**

I used ROS noetic and LiDAR os 128, but I noticed that I have the problem here:
/opt/ros/noetic/lib/nodelet/nodelet: line 1: 19849 Segmentation fault      (core dumped) $0 $@
================================================================================REQUIRED process [ouster/os_nodelet_mgr-1] has died!
process has died [pid 19794, exit code 139, cmd bash -c sleep 2; $0 $@ /opt/ros/noetic/lib/nodelet/nodelet manager __name:=os_nodelet_mgr __log:=/home/kws/.ros/log/7a244bd6-88b2-11ee-9ece-48b02d3d9056/ouster-os_nodelet_mgr-1.log].
log file: /home/kws/.ros/log/7a244bd6-88b2-11ee-9ece-48b02d3d9056/ouster-os_nodelet_mgr-1*.log
Initiating shutdown!
================================================================================
[relay_imu-6] killing on exit
[relay_lidar-5] killing on exit
[ouster/img_node-4] killing on exit
[ouster/os_cloud_node-3] killing on exit
[ouster/os_node-2] killing on exit
[ouster/os_nodelet_mgr-1] killing on exit
[ INFO] [1722953623.879997748]: Unloading nodelet /ouster/img_node from manager os_nodelet_mgr
[ INFO] [1722953623.881383688]: Unloading nodelet /ouster/os_cloud_node from manager os_nodelet_mgr
[ INFO] [1722953623.885051229]: waitForService: Service [/ouster/os_nodelet_mgr/unload_nodelet] could not connect to host [kws-desktop:41187], waiting...
[ WARN] [1722953623.885232479]: Couldn't find service os_nodelet_mgr/unload_nodelet, perhaps the manager is already shut down
[ INFO] [1722953623.885335969]: Unloading nodelet /ouster/os_node from manager os_nodelet_mgr
[ INFO] [1722953623.887873958]: waitForService: Service [/ouster/os_nodelet_mgr/unload_nodelet] could not connect to host [kws-desktop:41187], waiting...
[ WARN] [1722953623.888019048]: Couldn't find service os_nodelet_mgr/unload_nodelet, perhaps the manager is already shut down
[ INFO] [1722953623.888593648]: waitForService: Service [/ouster/os_nodelet_mgr/unload_nodelet] could not connect to host [kws-desktop:41187], waiting...
[ WARN] [1722953623.888743634]: Couldn't find service os_nodelet_mgr/unload_nodelet, perhaps the manager is already shut down

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. source ros environment
2. compile the project workspace
3. source the project workspace
4. ros launch ouster_ros sensor or replay
5. open another terminal and the project workspace
6. observe the issue ...

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
- Ouster Sensor? \[e.g. OS-0, OS-1, ..\]
- Ouster Firmware Version? \[e.g. 2.3, 2.4, ..\]
- ROS version/distro? \[e.g. noetic, foxy, humble, ..\]
- Operating System? \[e.g. Linux, Windows, MacOS\]
- Machine Architecture? \[e.g. x64, arm\]
- git commit hash (if not the latest).

Samahu, 2024-08-06:
This doesn't provide sufficient information that can help resolve the issue. Could you please enable GDB and attach the crash dump? Thanks

yulan0215, 2024-08-06:
Hi sorry for being late. Here:

process[relay_imu-6]: started with pid [32554]
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.2) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bash...
(No debugging symbols found in bash)
(gdb) GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.2) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bash...
(No debugging symbols found in bash)
(gdb) r[ INFO] [1722974117.125025772]: Loading nodelet /ouster/os_cloud_node of type nodelets_os/OusterCloud to manager os_nodelet_mgr with the following remappings:
[ INFO] [1722974117.138506039]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] could not connect to host [kws-desktop:54039], waiting...
n[ INFO] [1722974117.189501657]: Loading nodelet /ouster/img_node of type nodelets_os/OusterImage to manager os_nodelet_mgr with the following remappings:
[ INFO] [1722974117.201974528]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] could not connect to host [kws-desktop:54039], waiting...
[ WARN] [1722974127.139780812]: Waiting for service os_nodelet_mgr/load_nodelet to be available for 10 seconds...
[ INFO] [1722974127.144009305]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] could not connect to host [kws-desktop:54039], waiting...
[ WARN] [1722974127.208332197]: Waiting for service os_nodelet_mgr/load_nodelet to be available for 10 seconds...
[ INFO] [1722974127.215172283]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] could not connect to host [kws-desktop:54039], waiting...
[ WARN] [1722974137.155101938]: Waiting for service os_nodelet_mgr/load_nodelet to be available for 20 seconds...
[ INFO] [1722974137.159738945]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] could not connect to host [kws-desktop:54039], waiting...
[ WARN] [1722974137.231222920]: Waiting for service os_nodelet_mgr/load_nodelet to be available for 20 seconds...
[ INFO] [1722974137.234511379]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] could not connect to host [kws-desktop:54039], waiting...
[ WARN] [1722974147.180419064]: Waiting for service os_nodelet_mgr/load_nodelet to be available for 30 seconds...
[ INFO] [1722974147.186120393]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] could not connect to host [kws-desktop:54039], waiting...
[ WARN] [1722974147.245643133]: Waiting for service os_nodelet_mgr/load_nodelet to be available for 30 seconds...
[ INFO] [1722974147.248455525]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] could not connect to host [kws-desktop:54039], waiting...

Samahu, 2024-08-06:
Hi, you need to consult with https://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB. Make sure you build the code with `RelWithDebugInfo` otherwise you aren't likely to get a useful information.

also this seems like a very specific issue you are running into and not a problem in the driver so you may need to contact the technical support for this.