ouster-lidar/ouster-ros #352: "DROPPING PACKET" and "70% full, THROTTLING"

James-R-Han, 2024-08-06:
Hello!

I am using

- OS0-128 
- ROS2 Humble (with CycloneDDS) 
- UDP Profile RNG15_RFL8_NIR8
- Run configuration : `ros2 launch ouster_ros sensor.launch.xml sensor_hostname:=192.168.131.18 use_system_default_qos:=true timestamp_mode:=TIME_FROM_ROS_TIME sensor_qos_profile:=reliable proc_mask:="IMG|PCL" viz:=false`

I will experience the following errors:
[image]

Any advice on how to handle the situation?

Thank you in advance!

kavishmshah, 2024-08-07:
Hi, 
I too see the same when I have multiple subscribers to the pointcloud topic.
I'm using an nvidia Orin devkit and OS0-128U with 512x10, I don't see this issue occur but with 1024x10, I see packet dropa when Rviz is open and I record a bag simultaneously. 

What resolution are you using?

James-R-Han, 2024-08-07:
Hey Kavish! I'm glad I'm not the only one haha. I was using 1024x10 when I got the messages above

Samahu, 2024-08-08:
Hi @James-R-Han and @kavishmshah thanks for sharing the feedback.

I noticed that `sensor_qos_profile:=reliable` which suggest that you are not using the BEST_EFFORT which is what used for `SensorDataQoS`. I think choosing the RELIABLE QoS for the sensor will increase the holdup on the published queue resulting in the throttling and dropped packets. I recommend that you use the default `SensorDataQoS` for live processing and only use RELIABLE when capturing data.

In any case, as I noted in the current merged fix https://github.com/ouster-lidar/ouster-ros/pull/321 I have further TODOs that I want to implement soon which should improve the performance on the driver side.

So stay tuned :crossed_fingers:

James-R-Han, 2024-08-09:
Thanks @Samahu! 

When I've tried BEST_EFFORT publisher with BEST_EFFORT subscriber (RVIZ), empirically I see the frame rate will occasionally drop well below 10Hz.

When I use RELIABLE publisher with BEST_EFFORT subscriber, sometimes the RVIZ is just blank (see below). 

If I use RELIABLE for both I find the best result - the frame rate is consistently high; I'm able to wave my hand around and it's a smooth motion in the point cloud.

[image]

Samahu, 2024-08-09:
This problem could be tied to the underlying RMW used. I don't see this problem on x86 platforms with CycloneDDS. I didn't measure how good the driver works on NVIDIA devices but I do intend to once I get more free time. In any case, I think that [Zenoh](https://zenoh.io/) is official now and works with ROS2 which offers an alternative to the DDS communication layer and I have been hearing good reviews about it but I haven't tried myself. You can give it a try.

This is of course besides the fact that we can do more optimization on the driver side.

Limerzzz, 2024-08-22:
I face the same problem, but I fix it by set the proc_mask value as PCL in driver_params.yaml. And I drive it in python.

kavishmshah, 2024-08-22:
Hi @Samahu , @Limerzzz and @James-R-Han , 

With qos set to BEST_EFFORT, when parsing the ROS messages (after recording a bag file), we saw a lot of 'nan' values and decided to change it back to RELIABLE. Maybe this might explain the blank screen on RViz, not entirely sure though. 
Set use_system_default_qos: true, to have the system default settings, which is Reliable and Volatile. [Link](https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Quality-of-Service-Settings.html)

We were able to resolve the packet drop issue by doing the below:

1. Changing the DDS to cyclone and tuning it with multicast and localhost settings.
2. In addition to this, we also increased the memory buffer size.
3. Setting MTU to 9000 also helped.

We have tried the same on a NVIDIA Devkit and a high spec PC. In both the devices, we didn't notice packet drops after the change. I'll post in the commands/settings needed in a few days once we are able to replicate on another set of PCs, just to make sure. 

Thanks!

Samahu, 2024-09-16:
Hi @James-R-Han, I have implemented few improvements to the point cloud generation as part of #369 which I believe should help with your situation. This is partially to improving the handling of the function that generation the point cloud by allowing to skip copying the fields if the sensor didn't have valid returns for the specific pixel. Which should reduce the overhead. You can also reduce the effective range of the sensor via the two new launch file parameters `min_range` and `max_range` but this can depend on  your specific use case. Additionally users who may have limited bandwidth can switch to using non-organized point cloud by setting the `organized` parameter to false. I do have further improvements in the pipeline but this what I'd like to merge short term. Please let me know if you were able to check that the provided solution helps with your case.

outrider-jkoch, 2024-09-16:
If I'm not mistaken the driver collects a full scan and then batch processes it correct? If so, while it wouldn't save processing time could the driver instead do processing on each piece as it receives it? That way the work would split up into at most 128 (2048 / 16) small batches versus a single large batch. It might give CPUs a little bit more time to complete the required task. But if the CPU is heavily loaded this could still be problematic. Perhaps this has been tried before and not proven to be effective.

Samahu, 2024-09-17:
Regarding the `cartesian` being hefty I do have some planned optimizations that should significantly reduce its overhead but these will have to wait until the next or later release.

Samahu, 2024-10-08:
@James-R-Han Could you please try out the [last release](https://github.com/ouster-lidar/ouster-ros/releases/tag/ros2-v0.13.2) and let me know if it helps with your situation. Thanks.

James-R-Han, 2024-10-08:
Hi @Samahu! Thanks for implementing those improvements! Shortly after my post, I swapped to a better computer (better CPU) and the problem went away. So, I won't be able to explicitly recreate testing conditions when I first raised this post. Sorry about that!

Samahu, 2024-10-09:
Thanks, I do have a low end computer that exposed the same problem I will see if I can still re-produce before and after the change.

mariamercade, 2025-06-02:
> Have you switched to CycloneDDS as recommended by `ouster-ROS2/README.md`?

Yes, I already did everything using CycloneDDS, as recommended in the `ouster-ROS2/README.md`. The tests I mentioned were all done with CycloneDDS enabled.

mariamercade, 2025-06-03:
FYI: I just came across this related issue in Cyclone DDS: https://github.com/ros2/rmw_cyclonedds/issues/461. However, even with only one subscriber, the publish rate remains noticeably low for this ouster driver.

Samahu, 2025-06-03:
> I am using ROS2 jazzy as I mentioned befote

My bad .. but yeah it is mostly a ROS problem.. I don't have this issue when running on my laptop .. I can have multiple subscribers with no issues. Please make sure you are using the latest version of the driver because there was a problem at early time of the driver inception but has been resolved (checkout the [discussion](https://github.com/ouster-lidar/ouster-ros/issues/240#issuecomment-1816025774)).

mariamercade, 2025-06-03:
I am using ros2 branch and I don't see a recent commit with this fix (what you sent is only done for noetic, right?).

Samahu, 2025-06-03:
> I am using ros2 branch and I don't see a recent commit with this fix (what you sent is only done for noetic, right?).

They are linked to the same PR
[image]

kavishmshah, 2025-06-03:
> Hello again,
> 
> I have made further discoveries. Using Cyclone DDS and executing the following lines improved the rate to 10 Hz:
> 
> ```
> sudo sysctl -w net.core.rmem_max=16777216
> sudo sysctl -w net.core.rmem_default=16777216
> ```
> 
> However, when I have more than one subscriber to the pcl topic the rate drops and I have the same behavior as in the issue [#322](https://github.com/ouster-lidar/ouster-ros/issues/322) . I just came across this related issue in Cyclone DDS: [ros2/rmw_cyclonedds#461](https://github.com/ros2/rmw_cyclonedds/issues/461). I hope this helps!

Hey guys!

I had a similar issue and in addition to increasing the buffer, I tried the below:

1. **Setting local host env variable:** export ROS_LOCALHOST_ONLY=0
2. **Enabling multicast :** sudo ip link set lo multicast on
3. Cyclone DDS xml as in this [link](https://autowarefoundation.github.io/autoware-documentation/main/installation/additional-settings-for-developers/network-configuration/dds-settings/) and then set the DDS env variable as instructed in the link.

The above worked for me and didn't see packet drops.

PR #369 (2024-09-20): [HUMBLE|IRON|JAZZY] Port ROS-363 to ROS2
## Related Issues & PRs
- ported_from #363 
- related #10 
- related #152 
- related #175 
- resolves #238 
- resolves #251 
- resolves #259
- resolves #304
- related #306 
- resolves #352

## Summary of Changes
- Update to SDK client 0.11.1
- Switch to using LidarPacket and ImuPacket
  - Get's rid of the deprecation warnings
- Publish raw packets for the `os_driver`
- Add a flag to control the publish of organized point cloud vs non-organized
- Add a flag to control the publish of destaggerd point cloud vs staggered
- Add parameter to set the min/max l [... cut]