ouster-lidar/ouster-ros #365: REQUIRED process [ouster/os_nodelet_mgr-2] has died!

retepik, 2024-09-10:
Hi,
my configuration is :
Ubuntu 20.04
Ros : Noetic
Ouster OS-1 128 Rev C.
firmware: 2.5.3

after update to latest ouster-ros driver I get error

`REQUIRED process [ouster/os_nodelet_mgr-2] has died!
process has died [pid 43842, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=os_nodelet_mgr __log:=/home/mapper/.ros/log/3eee6cfc-6f8b-11ef-b8ff-4de79e617463/ouster-os_nodelet_mgr-2.log].
log file: /home/mapper/.ros/log/3eee6cfc-6f8b-11ef-b8ff-4de79e617463/ouster-os_nodelet_mgr-2*.log
Initiating shutdown!`

Strange think is that I get error only on Release build
`catkin_make --cmake-args -DCMAKE_BUILD_TYPE=Release`
when I use Debug, or RelWithDebInfo build type, all work.
I try it with fresh default installation of ROS and ouster-ros driver, but with same behavior

my command is :
`roslaunch --master-logger-level='debug' -v /home/mapper/Programs/ouster/ouster-ros/src/ouster-ros/launch/sensor.launch sensor_hostname:=169.254.192.176 udp_dest:=192.168.113.195 lidar_port:=7502 imu_port:=7503 replay:=false lidar_mode:=512x10 metadata:=/home/mapper/metadata.json viz:=true image:=true udp_profile_lidar:=LEGACY`

output : 
`... logging to /home/mapper/.ros/log/3eee6cfc-6f8b-11ef-b8ff-4de79e617463/roslaunch-EOS-PC-01-43815.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

... loading XML file [/opt/ros/noetic/etc/ros/roscore.xml]
... executing command param [rosversion roslaunch]
Added parameter [/rosversion]
... executing command param [rosversion -d]
Added parameter [/rosdistro]
Added core node of type [rosout/rosout] in namespace [/]
... loading XML file [/home/mapper/Programs/ouster/ouster-ros/src/ouster-ros/launch/sensor.launch]
Added node of type [nodelet/nodelet] in namespace [/ouster/]
Added parameter [/ouster/os_node/sensor_hostname]
Added parameter [/ouster/os_node/udp_dest]
Added parameter [/ouster/os_node/lidar_port]
Added parameter [/ouster/os_node/imu_port]
Added parameter [/ouster/os_node/udp_profile_lidar]
Added parameter [/ouster/os_node/lidar_mode]
Added parameter [/ouster/os_node/timestamp_mode]
Added parameter [/ouster/os_node/metadata]
Added parameter [/ouster/os_node/azimuth_window_start]
Added parameter [/ouster/os_node/azimuth_window_end]
Added parameter [/ouster/os_node/persist_config]
Added parameter [/ouster/os_node/attempt_reconnect]
Added parameter [/ouster/os_node/dormant_period_between_reconnects]
Added parameter [/ouster/os_node/max_failed_reconnect_attempts]
Added node of type [nodelet/nodelet] in namespace [/ouster/]
... loading XML file [/home/mapper/Programs/ouster/ouster-ros/src/ouster-ros/launch/common.launch]
Added parameter [/ouster/os_cloud_node/tf_prefix]
Added parameter [/ouster/os_cloud_node/sensor_frame]
Added parameter [/ouster/os_cloud_node/lidar_frame]
Added parameter [/ouster/os_cloud_node/imu_frame]
Added parameter [/ouster/os_cloud_node/point_cloud_frame]
Added parameter [/ouster/os_cloud_node/timestamp_mode]
Added parameter [/ouster/os_cloud_node/dynamic_transforms_broadcast]
Added parameter [/ouster/os_cloud_node/dynamic_transforms_broadcast_rate]
Added parameter [/ouster/os_cloud_node/proc_mask]
Added parameter [/ouster/os_cloud_node/scan_ring]
Added parameter [/ouster/os_cloud_node/ptp_utc_tai_offset]
Added parameter [/ouster/os_cloud_node/point_type]
Added node of type [nodelet/nodelet] in namespace [/ouster/]
Added parameter [/ouster/img_node/proc_mask]
Added node of type [nodelet/nodelet] in namespace [/ouster/]
Added node of type [rviz/rviz] in namespace [/]
... done importing include file [/home/mapper/Programs/ouster/ouster-ros/src/ouster-ros/launch/common.launch]
started roslaunch server http://EOS-PC-01:35957/
[... 95 lines cut ...]
Initiating shutdown!

[rviz-6] killing on exit
[ouster/img_node-5] killing on exit
[ouster/os_node-3] killing on exit
[ouster/os_cloud_node-4] killing on exit
[ouster/os_nodelet_mgr-2] killing on exit
[ INFO] [1725982949.473568623]: Unloading nodelet /ouster/img_node from manager os_nodelet_mgr
[ INFO] [1725982949.473692499]: Unloading nodelet /ouster/os_node from manager os_nodelet_mgr
[ INFO] [1725982949.473811439]: Unloading nodelet /ouster/os_cloud_node from manager os_nodelet_mgr
[ INFO] [1725982949.476741537]: waitForService: Service [/ouster/os_nodelet_mgr/unload_nodelet] could not connect to host [EOS-PC-01:47311], waiting...
[ WARN] [1725982949.476782891]: Couldn't find service os_nodelet_mgr/unload_nodelet, perhaps the manager is already shut down
[ INFO] [1725982949.477072606]: waitForService: Service [/ouster/os_nodelet_mgr/unload_nodelet] could not connect to host [EOS-PC-01:47311], waiting...
[ WARN] [1725982949.477121257]: Couldn't find service os_nodelet_mgr/unload_nodelet, perhaps the manager is already shut down
[ INFO] [1725982949.477417235]: waitForService: Service [/ouster/os_nodelet_mgr/unload_nodelet] could not connect to host [EOS-PC-01:47311], waiting...
[ WARN] [1725982949.477476105]: Couldn't find service os_nodelet_mgr/unload_nodelet, perhaps the manager is already shut down
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

Could you help me with this bug?
Thanks 
retepik

Samahu, 2024-09-10:
Hi, can you please indicate the driver version that you are currently using?

Samahu, 2024-09-10:
I have examined the log a bit and it seems you are rather using a recent
version of the driver (maybe with some modifications done to it), are you
able to reproduce the problem using original code of the driver with no
changes?

retepik, 2024-09-11:
Hi Samahu,
I didn't make changes in original code, I used cloned code from git repositary

yulan0215, 2024-10-17:
Did you have any solution for this? I have the same sensor and same problem, thx.

yulan0215, 2024-10-17:
> Hi Samahu, I didn't make changes in original code, I used cloned code from git repositary

Hi, did you have any solution for this? I have the same sensor and same problem, thx.

yulan0215, 2024-10-17:
I also used the very recently version ROS driver and the firmware is:
[image]
Can you please check if the problem was from firmware or IP?
The error is shown here:
[image]
Thanks for your help!

Samahu, 2024-10-17:
Here are few things to try that can help us narrow down the problem (the log isn't giving sufficient output unfortunately), try each step one by one:
* First update to the latest version 0.13.1
* Switch to `driver.launch` instead of the `sensor.launch` (it offers the same parameters).
* Try leaving `lidar_port` and `imu_port` values set to zero so the driver would pick random free ports
* Set `proc_mask="IMG"` alone or `proc_mask="IMU"` alone or even leave empty `proc_mask=""`, this should help us find if the bug is a result of one processors.

Finally, you may need to enable GDB (but let's try this after the steps suggested above).

Samahu, 2024-10-17:
> Can you please check if the problem was from firmware or IP?
> The error is shown here:

@yulan0215 your problem could be different, it is not clear how do you initialized the driver, a text log works would be much clearer than an image.

yulan0215, 2024-10-17:
> > Can you please check if the problem was from firmware or IP?
> 
> > The error is shown here:
> 
> 
> 
> @yulan0215 your problem could be different, it is not clear how do you initialized the driver, a text log works would be much clearer than an image. 

Many thanks for your quick reply. Although right now I could not change the Roslauch and I want to mention I also loss the data even there was no error or warning from terminal. I have to mention my ouster LiDAR directly connected to Jetson Xavier NX with Ethernet cable, do u have any idea for that? Many thx.

Samahu, 2024-10-17:
> I want to mention I also loss the data even there was no error or warning from terminal. I have to mention my ouster LiDAR directly connected to Jetson Xavier NX with Ethernet cable, do u have any idea for that? Many thx.

This might be a network related issue, try applying the recommend network settings described in https://github.com/ouster-lidar/ouster-ros/blob/master/util/network-configure.bash

yulan0215, 2024-10-19:
> This might be a network related issue, try applying the recommend network settings described in https://github.com/ouster-lidar/ouster-ros/blob/master/util/network-configure.bash

Hi, [image]
I have this issue of Ethernet problem. Do you think if this solution can solve the data loss problem? Thx

yulan0215, 2024-10-19:
> > > I want to mention I also loss the data even there was no error or warning from terminal. I have to mention my ouster LiDAR directly connected to Jetson Xavier NX with Ethernet cable, do u have any idea for that? Many thx.
> > 
> > 
> > This might be a network related issue, try applying the recommend network settings described in https://github.com/ouster-lidar/ouster-ros/blob/master/util/network-configure.bash
> 
> Hi, ![image](https://private-user-images.githubusercontent.com/62309959/378075118-2dd9166d-9589-4a17-a828-e58381289137.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkzNDQxNTQsIm5iZiI6MTcyOTM0Mzg1NCwicGF0aCI6Ii82MjMwOTk1OS8zNzgwNzUxMTgtMmRkOTE2NmQtOTU4OS00YTE3LWE4MjgtZTU4MzgxMjg5MTM3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE5VDEzMTczNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAzMDkzNjMxZjkyYzc2YmI4OTBhMTNiNmQ1ZTAyNzE3ZDMxYWQxMDUyZjkwMzlhNzQxMjU1MzVmMGRhMTMwYmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.eoZjRogykrSgf3ajx4LG-aJg-Na-k_3sh_jiULfx11Q) I have this issue of Ethernet problem. Do you think if this solution can solve the data loss problem? Thx
> 
> And I want to say that I had lidar data loss problem when the system ran long time.

## in addition you can try increasing the receive buffer size
# sudo ethtool -G <sensor_network_interface> rx 2048

## increase the MTU size to 9000 to decrease fragmentation
# sudo ifconfig <sensor_network_interface> mtu 9000 I used Jetson Xavier NX and do you have any suggestion if I need the configure above(commented) or not? If they are needed, do you have any suggested configuration?Like the 2048 is suitable in my case? Thx.

closed 2024-10-21 by Samahu

retepik, 2024-11-06:
I have still issue with this bug. I tested it on another PC with Ubuntu 20, ros noetic, with last git clone, with recommend network setting ([link](https://github.com/ouster-lidar/ouster-ros/blob/master/util/network-configure.bash)), with driver.launch, without setting imu and lidar port, but I have still same issue. I found that I have got issue on Release and  RelWithDebInfo when proc_mask contains PCL, another combination proc_mask without PCL work. In Debug work all combination proc_mask. 
I found also change ouput in console :
in Release : _waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] has not been advertised, waiting..._ 
in Debug: : _waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] could not connect to host_

Do you have any idea, how can I fix it? It's frustrating.
Thanks

Samahu, 2024-11-06:
I never tried tested the driver performance on NVIDIA NX, so it could be below the threshold that the driver with the default PointCloud size could run on, try using a lower lidar mode such as the 128x512.

retepik, 2024-11-07:
I do not use NVIDIA NX, I have standard desktop with AMD Ryzen 9 processor. I tested all lidar mode, but still same result.

Samahu, 2024-11-07:
Please contact [Ouster Support Staff](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86) so they can look closely at the problem.