ouster-lidar/ouster-ros #367: Problem applying driver_param when executing launch on jetson orin humble

lidarmansiwon, 2024-09-13:
**Describe the bug**
A clear and concise description of what the bug is.

I'm trying to use Ouster on Humble, Jetson Orin. Previously, when using it on foxy, the values ​​modified in driver_param were applied well and ouster was launched, but now on humble, only the values ​​modified in the ouster sensor host name ip by entering the sensor config are applied, and the values ​​in param are not applied at all. For example, I want to apply ROS TIME, but I can't.

Please help me.

Samahu, 2024-09-13:
Hi @lidarmansiwon, thanks for reporting this. What is the version of `ouster-ros` driver that you currently use? I have thoroughly tested the driver settings as part of this PR https://github.com/ouster-lidar/ouster-ros/pull/362 on Humble and settings were applied consistently.

lidarmansiwon, 2024-09-13:
I'm using ouster os2 and since it's humble, I made a git clone of the ros2 branch in this git repository. I built it according to the instructions below git. Is branch correct what driver means? Or does the ouster itself have a separate driver and need to update the firmware of the sensor itself? I've just done everything with git clone -bros2.

Samahu, 2024-09-13:
Can you provide me with the command you use to launch the driver?

lidarmansiwon, 2024-09-13:
Thank you very much for your quick response. I have stopped working on it due to personal reasons and will try again on September 18. If you don't mind, I will organize the detailed logs and commands in reply again then. Thank you again for your hard work.

lidarmansiwon, 2024-09-18:
Hello, let me summarize the problem and my situation.

[image]

The above is the page http://192.168.6.1/config. It is set up like that by default.

[image]

When I run ouster, the following happens. This is the same when I use the ''ros2 launch ouster_ros [driver.launch.py](http://driver.launch.py/) params_file:='' command.  It seems that the sensor host name is reflected among the parameters in the file, but the rest of the parameters are not reflected at all. UDP_dest, lidar_mode, timestamp_mode, etc. are not reflected at all, and only the sensor host name is reflected.

[image]
[image]

[image]

If I set udp_dest directly to my udp_dest, 192.168.6.1, in http://192.168.6.1/config, ouster runs without error, but I can't use TIME_FROM_ROS_TIME because it still doesn't reflect the remaining parameters.

Plese help me!

Samahu, 2024-09-18:
From the logs I see that you are getting `sensor::poll_client` right on startup, this is unusual and might suggest that you have a problem in the yaml configuration which results in the failure to connect, in this case ouster-ros will attempt to reconnect to the sensor, but in this case the node will retrieve the current configuration of the sensor might explain why you don't see the updated settings.

I see that you are using static ip for `udp_dest`, can I ask you to try and set this value to empty (i.e. `''` which then forces to generate the target ip automatically)

lidarmansiwon, 2024-09-21:
I sincerely appreciate your efforts. I am currently in a hurry and am using the docker foxy version. I will try again as per your advice as soon as possible. Thank you.

lidarmansiwon, 2024-09-24:
[image]
@Samahu The sensor keeps turning off and then back on, making a buzzing sound. However, I can't turn the sensor on.....

Samahu, 2024-10-03:
@lidarmansiwon sorry for the delayed response as I was on vacation, this might be caused by the bug reported on #376. I haven't yet tested the driver on Jetson Orion but still planning to. Meanwhile, can you update your driver to driver version v0.13.1 and check if this issue still persist (or not :crossed_fingers:). Thanks.

Samahu, 2024-10-17:
@lidarmansiwon I haven't heard back from you about for about two weeks. I do believe that #376 would fix the issue if you update to the current version. I am going to close the issue as resolve but let us know otherwise.

closed 2024-10-17 by Samahu

lemairelouis, 2024-11-19:
Hi, I’m commenting here because I’ve encountered the same issue and wanted to provide some additional details.

### Setup Description:
- ouster-ros v0.13.1
- Jetson Orin NX (running the driver in a ROS2 Humble Docker container)
- Lenovo Thinkbook laptop (for development and visualization)
- RUTX11 router (Ethernet connection with Jetson, Wi-Fi connection with laptop)

When I launch the ROS2 driver from the Jetson with an empty udp_dest, the driver starts correctly, but I’m unable to echo the /ouster/points topic. I suspect this is due to bandwidth limitations, so I tried directly sending UDP packets to my laptop. I expect this to work, because when launching the driver from my laptop, I am able to visualize /ouster/points (with echo and in rviz2).

When launching with "udp_dest=[laptop IP]", the configuration is updated in the Ouster WebUI and data packets are visible using Wireshark (from sensor IP to laptop IP) :

[image]

However, I get the "sensor::poll_client() returned error or timed out" error on driver startup, sensor restarts indefinitely and data is not published on topics.

Samahu, 2024-11-20:
Your "udp_dest" should point to the device where the driver is launched from. By manually setting this the sensor doesn't need to figure out the destination IP address.

The reason why you may not be getting the PointCloud2 published could be related to the underlying DDS middleware in use. Generally, we recommend CycloneDDS but I hadn't had the chance to check the performance on a Jetson Orin NX. So if you aren't using a CycloneDDS already I recommend that you give it a try.