ouster-lidar/ouster-ros #376: Why does the driver not reinitialize the sensor config when I pass parameters different from the currently active config?

themightyoarfish, 2024-10-02:
**Describe the bug**

When starting the ROS2 driver e.g. like this
```
ros2 launch ouster_ros sensor.launch.xml  sensor_hostname:=os-122307000738.local lidar_mode:=1024x10 viz:=false udp_profile_lidar:=RNG15_RFL8_NIR8
```

while the currently active lidar mode is 1024x20, the sensor config is not changed. The logging just says
```
[os_driver-1] [INFO] [1727862779.854131919] [ouster.os_driver]: Wrote sensor metadata to os-122307000738-metadata.json
[os_driver-1] [INFO] [1727862779.854166263] [ouster.os_driver]: ouster client version: 0.11.1+483206f-release
[os_driver-1] product: OS-1-128, sn: 122307000738, firmware rev: v3.0.1
[os_driver-1] lidar mode: 1024x20, lidar udp profile: RNG15_RFL8_NIR8
[os_driver-1] [INFO] [1727862779.855098502] [ouster.os_driver]: reset service created
[os_driver-1] [INFO] [1727862779.855713542] [ouster.os_driver]: get_metadata service created
```

What is the use for passing arguments if they do not modify the active config? Is there a way to change this and have the sensor reinitialize?

**To Reproduce**

1. set some lidar mode on via browser, set config active and persist
2. run driver (ros2 humble) with current lidar mode to verify function
3. stop and restart driver with different lidar mode
4. observe that new lidar mode is not configured

**Platform (please complete the following information):**
- Ouster Sensor? OS1-128 rev7
- Ouster Firmware Version? v3.0.1
- ROS version/distro? humble
- Operating System? ubuntu mate 22.04
- Machine Architecture? x64
- git commit hash (if not the latest): 5dd855558e6cf4d8408591803baaecae0ebad60d

themightyoarfish, 2024-10-02:
There seems to be no was to force the sensor to reinitialize with the chosen parameters, correct?

Samahu, 2024-10-03:
Hi @themightyoarfish, thanks for reporting the problem. I was able to re-produce the issue and I have created a fix for it. This was a result of a human error on my behalf when porting the reconnection logic from ROS1 driver to ROS2: https://github.com/ouster-lidar/ouster-ros/pull/369 and I haven't sufficiently tested the solution on the ROS2 branch. If you have time please take a look at https://github.com/ouster-lidar/ouster-ros/pull/380 and let me know if you still have a problem not getting the parameters applied on startup.

Samahu, 2024-10-03:
Fixed in 0.13.1 per #380

PR #380 (2024-10-03): HOTFIX/ROS-376-initialize-the-sensor-with-launch-config-params
## Related Issues & PRs
- closes #376
- closes #377
- closes #378

## Summary of Changes
- Make sure to initialize the sensor with specified launch file parameters
- Add a note recommending the use of CycloneDDS over FastDDS

## Validation
- launch ouster-ros with some new config and verify the new config is applied upon node startup
closed 2024-10-03 by Samahu