ouster-lidar/ouster-ros #377: Setting `proc_mask` once bricks the driver

themightyoarfish, 2024-10-02:
**Describe the bug**

I launch the driver node like this
```
ros2 launch ouster_ros sensor.launch.xml  sensor_hostname:=os-122307000738.local lidar_mode:=1024x20 viz:=false udp_profile_lidar:=RNG19_RFL8_SIG16_NIR16
```

And get some data on the `/ouster/points` topic at very low frequency (different issue…). If however, I stop the driver and add the `proc_mask` argument and restart with e.g. `proc_mask:=PCL`, no point clouds are being generated. If I then restart the driver with the default value of `proc_mask:='IMG|PCL|IMU|SCAN'`, still nothing. A computer reboot solved this last time, but I would like to know what the problem is.

**To Reproduce**
see above.

**Platform (please complete the following information):**
- Ouster Sensor? OS1-128 rev7
- Ouster Firmware Version? v3.0.1
- ROS version/distro? humble
- Operating System? ubuntu mate 22.04
- Machine Architecture? x64
- git commit hash (if not the latest): 5dd855558e6cf4d8408591803baaecae0ebad60d

themightyoarfish, 2024-10-02:
It seems that I could work around this by killing the shell/tmux session this was running in, and opening a new one with sourcing the ros and workspace setup files.

Samahu, 2024-10-03:
I am unable to re-produce this, probably resolved by #380

Samahu, 2024-10-03:
@themightyoarfish as I noted earlier I couldn't generate the problem. Could you confirm that the problem persists within 0.13.1 and maybe give more hints that help me re-produce the issue (if still exist). Thanks.

themightyoarfish, 2024-10-04:
I will try it again under controlled circumstances next week.

Samahu, 2024-10-08:
Could [this issue](https://github.com/ouster-lidar/ouster-ros/pull/384) be related?

themightyoarfish, 2024-10-11:
Can confirm that with current `ros2` head (fc911d1abd14bb1c2ca73b9b76dd57b8bfba45b4), this seems to not occur.

closed 2024-10-11 by themightyoarfish

Samahu, 2024-10-11:
awesome, thanks for the feedback.

PR #380 (2024-10-03): HOTFIX/ROS-376-initialize-the-sensor-with-launch-config-params
## Related Issues & PRs
- closes #376
- closes #377
- closes #378

## Summary of Changes
- Make sure to initialize the sensor with specified launch file parameters
- Add a note recommending the use of CycloneDDS over FastDDS

## Validation
- launch ouster-ros with some new config and verify the new config is applied upon node startup