ouster-lidar/ouster-ros #383: Not able to cooperate with Autoware under the namespace "sensing"

Michael-Mo0116, 2024-10-04:
**Describe the bug**
I installed the ouster driver and it works fine when I launch it individually. However, when I try to add it into the autoware launch file, the ouster will crash. After some research, I discovered that as long as I put the ouster launcher under the Autoware namespace "sensing", the ouster driver will crash. But it works fine outside that namespace. So, what is the problem with it and how can I solve it?

**Platform (please complete the following information):**
- Ouster Sensor: OS1-128
- Ouster Firmware Version: not sure
- ROS version/distro: ROS humble
- Operating System: Ubuntu 22.04
- Machine Architecture: x86_64
- git commit hash (if not the latest). I believe is from master branch?

Samahu, 2024-10-17:
Can you provide any logs that may help us resolve the issue? Thanks

Michael-Mo0116, 2024-10-22:
Sorry for the late reply. Here is the debug log when I'm running autoware with ouster:
```shell
auto2204@auto:~/ACRL/autoware$ ros2 launch autoware_launch autoware.launch.xml map_path:=$HOME/ACRL/map/test_map vehicle_model:=my_vehicle sensor_model:=my_sensor_kit --debug
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/auto2204/.ros/log/2024-10-19-15-46-07-998255-auto-211075
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7522b9db7ee0>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7522b9db7ee0>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7522b9dbc880>'
[DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 68, in try_get_launch_description_without_context
	perform_substitutions(context, self.__location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
	return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
	return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_context.py", line 240, in perform_substitution
	return substitution.perform(self)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/substitutions/launch_configuration.py", line 98, in perform
	raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: launch configuration 'vehicle_launch_pkg' does not exist

[DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: launch configuration 'vehicle_launch_pkg' does not exist
[DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 68, in try_get_launch_description_without_context
	perform_substitutions(context, self.__location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
	return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
	return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_context.py", line 240, in perform_substitution
	return substitution.perform(self)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/substitutions/launch_configuration.py", line 98, in perform
	raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: launch configuration 'sensor_launch_pkg' does not exist

[DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: launch configuration 'sensor_launch_pkg' does not exist
[DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 68, in try_get_launch_description_without_context
	perform_substitutions(context, self.__location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
	return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
	return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_context.py", line 240, in perform_substitution
	return substitution.perform(self)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/substitutions/launch_configuration.py", line 98, in perform
	raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: launch configuration 'directory_with_related_launches' does not exist

[DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: launch configuration 'directory_with_related_launches' does not exist
[DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_source.py", line 68, in try_get_launch_description_without_context
	perform_substitutions(context, self.__location)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
	return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
	return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_context.py", line 240, in perform_substitution
	return substitution.perform(self)
[... cut ...]
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7522b80a13c0>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x7522b7e62d10>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7522b80a13c0>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x7522b7e62a70>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7522b80ae7a0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7522b80ae7a0>' ✓ '<launch.event_h

Samahu, 2024-10-23:
Based on a brief look to the log I can see you have a problem with your robot setup. It appears that `base_link` is declared in more than place but you may have already had this issue.

Regarding the ouster ROS driver itself, I can't pinpoint the exact problem (it is not clear if the ouster-ros log is written to the console window or a separate file). But I can suggest two that you can try (before we delve deeper into this):
* Set the launch parameter `organized` on the ROS node to `false`, some packages doesn't handle NaN(s) values so this would omit these values from the generated point cloud.
* You may want to rename the driver tf transforms or turn them off completely if you are re-defining them elsewhere in your setup. You can rename the transforms directly via the parameters file, but you can't turn them off without modify the code at the moment.

Michael-Mo0116, 2024-10-28:
I'm sorry. But I think we are using the branch ros2-humble-sparse-neighbor-culling, which seems doesn't have the launch parameter `organized`. Should we change our drive to master branch? Also, it seems that we are also encountering a issue like pointcloud layout compatablility. It seems that autoware is using `PointXYZIRCAEDT` and our ouster driver is using `PointXYZIRC`. Can changing to master branch also solve that problem or we need another way to solve it?

Samahu, 2024-10-29:
@Michael-Mo0116 does your work depend on the prototyped `sparse-neighbor-culling`? This was implemented as a quick prototype and not intended for production. But it will be integrated into the mainline eventually. If your work doesn't need the `sparse-neighbor-culling` filter I would recommend that you switch to the `ros2` branch (the mainline for ros2).

Regarding the support for `PointXYZIRCAEDT`, this was brought up by others https://github.com/ouster-lidar/ouster-ros/issues/217 and you can see my notes on it. It definitely further work from us to support all the fields of `PointXYZIRCAEDT` but if you merely creating the `PointXYZIRCAEDT` type and filling non-supported fields with zeros then this [PR](#290) - which I intend to integrate soon - gives you the steps needed to adapt the sensor native format to the `PointXYZIRCAEDT` format