ouster-lidar/ouster-ros #389: replay_pcap is not working with cyclone_dds.

jaan1729, 2024-10-17:
> **IMPORTANT**  
> Before reporting a new bug make sure you have:
> - [x] Checked existing issues https://github.com/ouster-lidar/ouster-ros/issues
> - [x] Checked existing discussions https://github.com/ouster-lidar/ouster-ros/discussions
> - [x] Checked ouster community https://community.ouster.com/
> If you couldn't find relevant information and you think this is rather a driver 
> problem then proceed with bug reporting.

**Describe the bug**
The ouster_pcap node is not working with CycloneDDS. However, it works fine when the DDS is set to default.

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. Setting the DDS: 
`export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`
2. compile the project workspace:
`colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DBUILD_PCAP=ON --packages-select ouster_sensor_msgs ouster_ros`
3. ros launch ouster_ros sensor or replay:
`ros2 launch ouster_ros replay_pcap.launch.xml pcap_file:=/workspace/pcaps/2024099_1400_OS-1-128_122421001099.pcap metadata:=/workspace/pcaps/2024099_1400_OS-1-128_122421001099.json`
4. The Issue 
[image]

5. Setting default DDS:
`export RMW_IMPLEMENTATION=''`
6. ros launch ouster_ros sensor or replay:
`ros2 launch ouster_ros  replay_pcap.launch.xml pcap_file:=/root/ros2_bags/pcaps/2024099_1400_OS-1-128_122421001099.pcap metadata:=/root/ros2_bags/pcaps/2024099_1400_OS-1-128_122421001099.json`
7. Working successfully: 
[image]

**Platform (please complete the following information):**
- Ouster Sensor: OS-1
- ROS distro: humble
- Operating System: Linux 22.04
- Docker Environment: Yes
- Machine Architecture: x64
- git commit hash (ros2-v0.13.2).

Samahu, 2024-10-17:
@jaan1729 this appears to be a timing issue, the replay launch file which are currently implemented in xml format can't use the more proper event based activation. So it currently configures and activates the pcap replay node after a fixed time has passed, which seems to be insufficient when using CycloneDDS, I think the easiest workaround probably to increase the sleep period on the two following execute commands within the `pcap_replay.launch.xml`:
```xml
  <executable if="$(var _use_metadata_file)"
    cmd="$(find-exec ros2) lifecycle set /$(var ouster_ns)/os_pcap configure"
    launch-prefix="bash -c 'sleep 0; $0 $@'" output="screen"/>
  <executable if="$(var _use_metadata_file)"
    cmd="$(find-exec ros2) lifecycle set /$(var ouster_ns)/os_pcap activate"
    launch-prefix="bash -c 'sleep 1; $0 $@'" output="screen"/>
```
This is a workaround to get you going, ultimately a better solution would be to port these launch files to the python format or implement the "auto_start" flag similar to the sensor/driver nodes.

Hope this helps!

jaan1729, 2024-10-18:
I tried increasing the sleep duration in the launch file (up to 20 seconds) but no success.

I commented out the lifecycle commands within the launch file and re-ran it. This resulted in the following log message

[image]

Despite the launch file apparently functioning with commented-out lifecycle commands, I can't find `/ouster/os_pcap` node from both `ros2 lifecycle nodes` and `ros2 node list` commands. Both the launch file and the current environment are configured with the same RMW implementation (rmw_cyclonedds_cpp)

Interestingly, I can find the node with the same steps when using the default DDS implementation.

Samahu, 2024-10-18:
I can't re-produce the same behavior on my machine after commenting out the block above. The `/ouster/os_pcap` does show up in both lists (I use CycloneDDS on Ubuntu 22.04 running Humble). Maybe consider applying the latest updates on your system and see if that helps.

Samahu, 2024-10-29:
@jaan1729 I have made changes to how the pcap node get's initialized which avoid the use of timed requests to start the replay. If you have time please take a look at #393 and let me know if resolves this issue.

Samahu, 2024-10-30:
Resolved by #393

PR #393 (2024-10-30): ROS-389 [rolling/humble/iron/jazzy]: replay-improvments-and-fixes
## Related Issues & PRs
- related: #392
- resolves: #153
- resolves #389
- integrates: #330

## Summary of Changes
* Remap `/metadata` topic
* Support **loop** capability in pcap replay
* Add `play_delay` to regular replay and pcap replay
* Expose `play_rate` for the bag file replay
* Added a parameter `pub_static_tf` to disable tf tansforms broadcast 

## Validation
* Test the newly exposed flags for `replay_pcap`
```bash
ros2 launch ouster_ros replay_pcap.launch.xml pcap_file:=<...> json_file:=<...> loop:=true play_delay:=10
` [... cut]
closed 2024-10-30 by Samahu