ouster-lidar/ouster-ros #390: Failed to initialize client

Nour-BCH, 2024-10-18:
I'm trying to connect Ouster OS0-32-U Lidar with ROS Noetic and ubuntu 20.04, using the v2.5.3 FW with link-local only Ethernal interface

When trying to launch driver.launch I get the error:
[ouster::sensor] {error] init_client(): A timeout occurred while waiting for the sensor to initialize.
[1729259432.079667441]: Failed to initialize client

Unable to display sensor data in rviz, and no messages sent on the sensor topics

Samahu, 2024-10-18:
Hi @Nour-BCH, I can't reproduce the error with a REV-6 (128) sensor running FW 2.5.3. It is not clear to me how are you launching the sensor (Whether you are using the sensor hostname or the IP). Based on the error output this suggest that the sensor URL is probably un-accessible from the machine you are spawning the driver on. Consider the following:
* Make sure you are able to ping the sensor from the machine (whether you are using the IP address or the sensor hostname)
* From the same machine check if the command `curl -i -X GET http://<use-sensor-address>/api/v1/sensor/metadata` returns a valid json output. If it doesn't then this is likely a problem in your setup.
* If you are using the sensor hostname try out the sensor ip address. Sometime it could be a DNS problem where the system can't resolve the hostname of the sensor

Hope this helps!

Nour-BCH, 2024-10-21:
the connection with REV D sensor fails when launching the sensor using both sensor hostname and IP address.

I am able to ping the sensor from the machine

the curl -i -X GET http://<use-sensor-address>/api/v1/sensor/metadata command returns a valid json output

I don't know what can be the problem with my setup

Samahu, 2024-10-21:
I haven't tested the driver against a `REV D` a while back, let me see if I can grab one and try to re-produce the issue.
Meanwhile you can try the versions 0.5.0 and 0.4.0 of the driver and see if it gives you a better outcome.

Samahu, 2024-10-21:
@Nour-BCH I have just tested the driver against a REV D sensor with FW 2.5.3. Unfortunately, I can't re-produce the issue. It might be the sensor isn't functioning properly or there is a problem on your setup. Here are some stuff that you could try before contacting Ouster support:
- Check the Sensor Diagnostics page for any active alerts.
- Connect directly to the sensor i.e no switch if you were using one. Try a different machine, ...
Otherwise, please contact [Ouster support](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86)

Closing the issue as this isn't a driver problem.

closed 2024-10-21 by Samahu