ouster-lidar/ouster-ros #391: [ros2] build issues  ros2-v0.13.2+

juliangaal, 2024-10-21:
## IMPORTANT
> Before reporting a new bug make sure you have:
- [x] Checked existing [issues](https://github.com/ouster-lidar/ouster-ros/issues), none relevant to geometry_msgs or FlatBuffers
- [x] Checked existing [discussions](https://github.com/ouster-lidar/ouster-ros/discussions)
- [x] Checked ouster [community](https://community.ouster.com/)
> If you couldn't find relevant information and you think this is rather a driver 
> problem then proceed with bug reporting.

**Describe the bug**
* since ros2-v0.13.2, docker build fails with missing `geometry_msgs` Header
* since ros2 HEAD (`fc911d1abd14bb1c2ca73b9b76dd57b8bfba45b4`), the SDK has to be compiled without OSF to avoid Flatbuffer error 

## To Reproduce
Steps to reproduce the behavior (steps below are just an example):

```
git clone --depth 1 --branch ros2-v0.13.2 --recursive git@github.com:ouster-lidar/ouster-ros.git && cd ouster-ros
docker build --progress=plain --build-arg ROS_DISTRO=humble -t ouster .
```

Fails with

```
#12 22.96 /opt/ros/humble/include/tf2/tf2/buffer_core.h:49:10: fatal error: geometry_msgs/msg/velocity_stamped.hpp: No such file or directory
#12 22.96    49 | #include "geometry_msgs/msg/velocity_stamped.hpp"
```

(temporary) fix: Adding `ros-${ROS_DISTRO}-geometry_msgs` to apt installation step

```
&& apt-get install -y       \
    build-essential         \
    cmake                   \
    fakeroot                \
    dpkg-dev                \
    debhelper               \
    python3-rosdep          \
    python3-rospkg          \
    python3-bloom           \
    ros2-${ROS_DISTRO}-geometry-msgs \          <===========
    python3-colcon-common-extensions
```

this fix makes little sense to me, because, IMO, the dependencies are correctly stated

https://github.com/ouster-lidar/ouster-ros/blob/fc911d1abd14bb1c2ca73b9b76dd57b8bfba45b4/ouster-ros/package.xml#L19

https://github.com/ouster-lidar/ouster-ros/blob/fc911d1abd14bb1c2ca73b9b76dd57b8bfba45b4/ouster-ros/CMakeLists.txt#L72

https://github.com/ouster-lidar/ouster-ros/blob/fc911d1abd14bb1c2ca73b9b76dd57b8bfba45b4/ouster-ros/CMakeLists.txt#L112

### Discussion

* I don't quite understand why these issues weren't caught in CI, when I does essentially the same?
  
  https://github.com/ouster-lidar/ouster-ros/blob/fc911d1abd14bb1c2ca73b9b76dd57b8bfba45b4/.github/workflows/docker-image.yml#L28-L31  
* This seems like a rosdep bug? Am I doing something wrong?

### Environment

- ROS version/distro? humble
- Operating System? Linux
- Docker version? 27.3.1
- Machine Architecture? x64
- git commit hash? ros2-v0.13.2 and ros2 HEAD

Thanks for your help!

Samahu, 2024-10-21:
I am unable to reproduce the problem given the steps following the exact steps that you have provided.

closed 2024-10-22 by Samahu

juliangaal, 2024-10-22:
still having the issue on my local system, but not in gitlab ci! Thanks for you help, this must be an issue local to my docker