ouster-lidar/ouster-ros #395: Need help for Voxel Layer local costmap

Cavalletta98, 2024-11-05:
**Describe the bug**
I'm trying to use the PointCloud from the lidar inside the voxel layer of the local costamp but it is empty. The lidar is mounted on a clearpath jackal robot

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):

1. In one terminal on the robot: ros2 launch clearpath_nav2_demos nav2.launch.py use_sim_time:=false
2. In another terminal on the robot: ros2 launch clearpath_nav2_demos slam.launch.py use_sim_time:=false
3. On the host machine: ros2 launch clearpath_viz view_navigation.launch.py namespace:=j100_0538

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
- Ouster Sensor?OS1-32-U
- ROS version/distro?Humble
- Operating System? Ubuntu 22.04
- Machine Architecture? x64
- git commit hash 2a0ef50f1dbe3f512321e809d6ca87391ce38f6e

Attached you can find the yaml files for nav2 and slam
[params.zip](https://github.com/user-attachments/files/17629220/params.zip)

Samahu, 2024-11-11:
Are you able to receive point cloud using the official ouster-ros driver?

Cavalletta98, 2024-11-12:
It's seem to be a problem with the CPU usage going to 95% when i run all Nav2 stuffs. Now i move all the Nav2 stuffs outside the robot pc. I will let you know if it will fix

Samahu, 2024-11-20:
Does the problem still exist?