ouster-lidar/ouster-ros #397: Help me understand the PointField data

AndreasGronlund, 2024-11-07:
Confused by the offsets of the pointfields. Nut sure it is a bug, proably me not calculating correctly since the code works, but shouldn't the t field have offset 12 instead of 16. Or is z value written as float64 but read as float32 at the reciever's end making a gap of extra 4 bits to the t value. Se figure.

[image]

closed 2024-11-07 by Samahu

AndreasGronlund, 2024-11-28:
@Samahu does it mean that the PointCloud2 messages generated from the ros2 driver using the PCL-format generate more data and take up more bandwidth compared to a method of writing just the raw data of the x,y,z,...etc values? From my calculations the below message should have a point_step of (x, y, z, intensity, t, reflectivity, ring, ambient, range): 4+4+4+4+4+2+2+2+4=30 bytes just counting the data transmitted. The point instead has a point_step of 48 bytes which is 60% more than 30 bytes which I assume is due to padding from the PCL-data point type?

Regards,

/Andreas

[image]

Samahu, 2024-12-03:
You calculation is not accounting for padding. This is not something specific about the ouster-ros but rather any PCL point type has some padding. If your concern is mainly with the bandwidth consider using a different point type per: https://github.com/ouster-lidar/ouster-ros/discussions/260

AndreasGronlund, 2024-12-04:
> You calculation is not accounting for padding. This is not something specific about the ouster-ros but rather any PCL point type has some padding. If your concern is mainly with the bandwidth consider using a different point type per: #260

I understand the padding and that it is, at least partly, due to the PCL implementation and the struct of the data. I just think it seems excessive that 37.5% (30 -> 48bytes) of the data transmitted is padding. I have used the different data point types to reduce the rate from the lidar, so yes I know about it. But, does any of the selections below avoid or reduce the padding and if so which of them? Or is it just a reduection of fields? 

- original: This uses the original point representation ouster_ros::Point of the ouster-ros driver.
- native: directly maps all fields as published by the sensor to an equivalent point cloud representation with the additon of ring
- and timestamp fields.
- xyz: the simplest point type, only has {x, y, z}
- xyzi: same as xyz point type but adds intensity (signal) field.
- xyzir: same as xyzi type but adds ring (channel) field. This type is same as Velodyne point cloud type.

Could the struct with the point data be altered or avoided to reduce the padding? 

1.  E.g. avoiding the point struct by storing the all the fields data in an array of floats, sort of already searialized before the memcpy to pointcloud.data. This would increase the actual data with 3*2 = 6 bytes per point (reflectivity, ring and ambient), but reduce padding to zero. So just 
```
sensor_msgs::msg::PointCloud2 pointCloud;
int offset = 0;
for(int i = 0 ; i < numFields; i++){
      offset = addPointField(pointCloud, fieldNames[i], 1, sensor_msgs::msg::PointField::FLOAT32, offset);
}
pointCloud.point_step = offset;
```
and copying the data to the message
```
std::memcpy(&pointCloud.data[0], &lidarData[0], pointCloud.height * pointCloud.row_step);
```
where the lidarData is the array containing the sensor data and the PointCloud is the, to be published, PointCloud.

2. Or maybe the struct arranged in some different order, see e.g. this comment https://github.com/PointCloudLibrary/pcl/issues/4939#issuecomment-1048454840

Best regards,

/Andreas