ouster-lidar/ouster-ros #400: ROS2 Humble colcon build failed

tenmoonsninesky, 2024-11-14:
Ubuntu 22.04+ROS Humble

I have tried updating ROS2, but the compilation still failed.

(base) guangdian@guangdian-Default-string:~/OSROS2$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> ouster_sensor_msgs
Finished <<< ouster_sensor_msgs [6.00s]                     
Starting >>> ouster_ros
[Processing: ouster_ros]                             
[Processing: ouster_ros]                                     
--- stderr: ouster_ros                                          
CMake Warning at CMakeLists.txt:65 (add_library):
  Cannot generate a safe runtime search path for target ouster_ros_library
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libcurl.so.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/guangdian/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:98 (add_library):
  Cannot generate a safe runtime search path for target os_sensor_component
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libcurl.so.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/guangdian/anaconda3/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:125 (create_ros2_component)

CMake Warning at CMakeLists.txt:98 (add_library):
  Cannot generate a safe runtime search path for target os_replay_component
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libcurl.so.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/guangdian/anaconda3/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:135 (create_ros2_component)

CMake Warning at CMakeLists.txt:98 (add_library):
  Cannot generate a safe runtime search path for target os_cloud_component
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libcurl.so.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/guangdian/anaconda3/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:145 (create_ros2_component)

CMake Warning at CMakeLists.txt:98 (add_library):
  Cannot generate a safe runtime search path for target os_image_component
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libcurl.so.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/guangdian/anaconda3/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:155 (create_ros2_component)

CMake Warning at CMakeLists.txt:98 (add_library):
[... 54 lines cut ...]
client.cpp:(.text+0x4d2b): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)'
/usr/bin/ld: ouster_example/ouster_client/libouster_client.a(client.cpp.o): in function `ouster::sensor::init_client(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ouster::sensor::lidar_mode, ouster::sensor::timestamp_mode, int, int, int, bool)':
client.cpp:(.text+0x5ce2): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)'
/usr/bin/ld: ouster_example/ouster_client/libouster_client.a(client.cpp.o): in function `ouster::sensor::mtp_init_client(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ouster::sensor::sensor_config const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, int, bool)':
client.cpp:(.text+0x643a): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)'
/usr/bin/ld: ouster_example/ouster_client/libouster_client.a(client.cpp.o): in function `ouster::sensor::get_metadata[abi:cxx11](ouster::sensor::client&, int, bool)':
client.cpp:(.text+0x6e0b): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)'
/usr/bin/ld: ouster_example/ouster_client/libouster_client.a(client.cpp.o):client.cpp:(.text+0x6e78): more undefined references to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)' follow
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/ouster_ros_test.dir/build.make:346: ouster_ros_test] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:537: CMakeFiles/ouster_ros_test.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:166: all] Error 2
---
Failed   <<< ouster_ros [1min 25s, exited with code 2]

Summary: 1 package finished [1min 31s]
  1 package failed: ouster_ros
  1 package had stderr output: ouster_ros

**Platform (please complete the following information):**
- Ouster Sensor? \OS1-128-
- ROS version/distro? \humble
- Operating System? \Linux
- Machine Architecture? \x64

Samahu, 2024-11-14:
The build problem is mostly specific to your own setup. It seems that you have conflicting package libfmt. The default for your Ubuntu 22.04 should 8.x however somehow you have version 9.x. So you need to resolve this package conflict or allow them to exist side by side. The problem is not relevant to the driver code.

closed 2024-11-14 by Samahu

tenmoonsninesky, 2024-11-18:
> I don't know what's going on on your setup but it is clear that you have some problems when linking spdlog. libfmt is used by spdlog and the conflict between libfmt versions seems to affect the build. Probably try uninstalling all libfmt and libspdlog from your system then re-install `libspdlog-dev` and see if that fixes it.

Hello, when I tried to uninstall libfmt and libspdlog, ROS Humble was also uninstalled, so I reinstalled ROS. However, I found that installing ROS comes with installing libfmt and libspdlog, which are the same version as before my first uninstallation. Compiling ouster-ros will also result in the same error.

Here are the version of my libfmt and libspdlog:

```
libfmt-dev/jammy,now 8.1.1+ds1-2 amd64 [installed,automatic]
libfmt8/jammy,now 8.1.1+ds1-2 amd64 [installed,automatic]
libspdlog-dev/jammy,now 1:1.9.2+ds-0.2 amd64 [installed,automatic]
libspdlog1/jammy,now 1:1.9.2+ds-0.2 amd64 [installed,automatic]
ros-humble-rcl-logging-spdlog/jammy,now 2.3.1-1jammy.20240728.205441 amd64 [installed,automatic]
ros-humble-spdlog-vendor/jammy,now 1.3.1-1jammy.20240728.202943 amd64 [installed,automatic]
```

Is it because Ouster-ros does not support the latest version of ROS Humble?

Samahu, 2024-11-18:
It does support: https://github.com/ouster-lidar/ouster-ros/actions/runs/11601391393/job/32304005831

Samahu, 2024-11-18:
> Here are the version of my libfmt and libspdlog

I have the same versions on my machine, I am using ROS2 Humble mainly. Are you still getting the same error? Or is it a different error?

> It does support: https://github.com/ouster-lidar/ouster-ros/actions/runs/11601391393/job/32304005831

As above link shows the driver builds correctly under ROS2 Humble. The issue is specific to your own setup. If you are still unable to fix the build please bring this up with [our technical support team](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86).

tenmoonsninesky, 2024-11-19:
> > Here are the version of my libfmt and libspdlog
> 
> I have the same versions on my machine, I am using ROS2 Humble mainly. Are you still getting the same error? Or is it a different error?

Yes, I still have the same error.

```
/usr/bin/ld: warning: libfmt.so.8, needed by /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2, may conflict with libfmt.so.9
/usr/bin/ld: ouster_example/ouster_client/libouster_client.a(client.cpp.o): in function `ouster::sensor::(anonymous namespace)::mtp_data_socket(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
client.cpp:(.text+0x420b): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)'
/usr/bin/ld: ouster_example/ouster_client/libouster_client.a(client.cpp.o): in function `ouster::sensor::(anonymous namespace)::udp_data_socket(int)':
client.cpp:(.text+0x4d2b): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)'
/usr/bin/ld: ouster_example/ouster_client/libouster_client.a(client.cpp.o): in function `ouster::sensor::init_client(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ouster::sensor::lidar_mode, ouster::sensor::timestamp_mode, int, int, int, bool)':
client.cpp:(.text+0x5ce2): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)'
/usr/bin/ld: ouster_example/ouster_client/libouster_client.a(client.cpp.o): in function `ouster::sensor::mtp_init_client(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ouster::sensor::sensor_config const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, int, bool)':
client.cpp:(.text+0x643a): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)'
/usr/bin/ld: ouster_example/ouster_client/libouster_client.a(client.cpp.o): in function `ouster::sensor::get_metadata[abi:cxx11](ouster::sensor::client&, int, bool)':
client.cpp:(.text+0x6e0b): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)'
/usr/bin/ld: ouster_example/ouster_client/libouster_client.a(client.cpp.o):client.cpp:(.text+0x6e78): more undefined references to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v9::basic_string_view<char>, spdlog::level::level_enum, fmt::v9::basic_string_view<char>)' follow
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/ouster_ros_test.dir/build.make:346: ouster_ros_test] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:537: CMakeFiles/ouster_ros_test.dir/all] Error 2
gmake: *** [Makefile:166: all] Error 2
---
Failed   <<< ouster_ros [1.44s, exited with code 2]

Summary: 1 package finished [3.74s]
  1 package failed: ouster_ros
  1 package had stderr output: ouster_ros

tenmoonsninesky, 2024-11-20:
> I have the same versions on my machine, I am using ROS2 Humble mainly. Are you still getting the same error? Or is it a different error?
>
>
> As above link shows the driver builds correctly under ROS2 [...]

Hello, I just fixed this. According to the assistance provided by Ouster's engineers, this seems to be an error caused by not closing Anaconda.

Run the command below to deactivate Anaconda:
```
conda deactivate
```

Samahu, 2024-11-20:
awesome, thanks for providing the feedback, this might help others with similar situation.