ouster-lidar/ouster-ros #405: Script for post processing of rosbag to convert packets to pointcloud and imu messages

nkhedekar, 2024-11-20:
**Is your feature request related to a problem? Please describe.**
When working on a computationally constrained system (eg. a single board computer like a raspberry pi) with a full stack running, it is necessary to only record the raw packets from the lidar in the rosbag. If one needs the pointcloud at a certain time from this rosbag at a later point, it is necessary to use the replay launch file and play the bag. So if you were to share the rosbag with someone, that person needs to also have this repository built and functional - they cannot just use the rosbag. 

**Describe the solution you'd like**
Instead, if there were a script/node that rewrites the rosbag to also have the pointcloud and imu topics, this would allow the bag to be used independent of this repository. This would make it easier to share rosbags without additional dependance. 

The proposed new process would be:
1. Collect your rosbag as you currently do on computationally constrained systems with just the packets. 
2. **After collection, run a script to read the bag and rewrite it with the converted pointcloud and imu topics**
3. Distribute/use the bag without needing to run replay.launch

**Describe alternatives you've considered**
The bag can be played back with the replay.launch and you could re-record a new bag to have the pointcloud and imu topics but this is inefficient, dependant on your system and will change recording times in the new rosbag.

**Targeted Platform (please complete the following information only if applicable, otherwise dot N/A):**
- ROS version/distro: noetic

Samahu, 2024-11-20:
I understand it is desirable to work with the processed point clouds but couldn't one achieve this using the existing tools. I mean you can simply replay the bag file containing the raw packets and then run rosbag utility in a separate terminal to save the processed Imu & PointCloud messages:

```bash
roslaunch ouster_ros replay.launch bag_file:=<bag file path>
```
in a separate terminal
```bash
rosbag record -O /ouster/imu /ouster/points /ouster/points2
```

In the future I am considering to add an option to the record to allow users save the processed topics directly but give that this is easily achievable as described above this would be a low priority item.

nkhedekar, 2024-11-20:
> **Describe alternatives you've considered** The bag can be played back with the replay.launch and you could re-record a new bag to have the pointcloud and imu topics but this is inefficient, dependant on your system and will change recording times in the new rosbag.

In cases where time of arrival matters this would rewrite the time depending on the system load etc when re-recording making it meaningless. Additionally, this is not scalable with the duration of the bag (say you have a bag that is 4 hours long - you would need to spend 4 more hours to re-record this. You could play it back at a higher rate, but then you might inadvertently miss a packet somewhere). A cleaner and more correct solution is to just have a script that will do this for you correctly and as fast as possible by your system. 

The low priority is understandable and the current tools are a solution - just not a good one for our situation :sweat_smile: