ouster-lidar/ouster-ros #407: How can I compute the (relative) timestamp of each point?

themightyoarfish, 2024-11-27:
**Describe your question**

Each measurement block appears to have a timestamp as documented [here](https://static.ouster.dev/sensor-docs/image_route1/image_route3/sensor_data/sensor-data.html#lidar-data-packet-format), but I couldn't quickly find how points are associated to blocks exactly.

What arithmetic do I have to perform to get the timestamp for each point? There is code [here](https://github.com/ouster-lidar/ouster_example/blob/13ea8e8b8a4951fb630dbc9108666995c8443bf6/ouster_ros/src/ros.cpp?_pjax=%23js-repo-pjax-container%2C%20div%5Bitemtype%3D%22http%3A%2F%2Fschema.org%2FSoftwareSourceCode%22%5D%20main%2C%20%5Bdata-pjax-container%5D#L126) which simply accesses the `LidarScan` header for each column of the data, does this mean that each block contains one column? And is this the definitive way to do it? This wasn't obvious to me from when this question got [asked previously](https://github.com/ouster-lidar/ouster_example/issues/184).

Samahu, 2024-11-27:
> The lidar scan timestmap in 2048x10 mode has values ranging from 3771658217864 for column 1 to 3771758155224 for column 2047. While the delta is the expected 100ms, the absolute values do not seem to make sense. Should the first column not have a timestamp of 0 ns?

Are you talking about the timestamps reported in ouster-ros PointCloud? They should be relative and the timestamp for the first value should be 0.

Samahu, 2024-11-27:
> The lidar scan timestmap in 2048x10 mode has values ranging from 3771658217864 for column 1 to 3771758155224 for column 2047. While the delta is the expected 100ms, the absolute values do not seem to make sense. Should the first column not have a timestamp of 0 ns?
This column timestamp is relative to the internal sensor clock, right? In that case, should i subtract the extrapolated or real first column's timestamp to get to "ns since rotation start"? I copied the code from ouster-ros, and it does not seem to do this when system time mode is used. Hence I'm a bit confused how the condition in
[ouster-ros/src/point_cloud_compose.h](https://github.com/ouster-lidar/ouster-ros/blob/9711abd69834972a675e4a8ffad261231b021364/src/point_cloud_compose.h#L158)

>Line 158 in [9711abd](https://github.com/ouster-lidar/ouster-ros/commit/9711abd69834972a675e4a8ffad261231b021364)
 timestamp[ts_idx] > scan_ts ? timestamp[ts_idx] - scan_ts : 0UL; 
doesnt always lead to zero timestamps because timestamp[idx] is relative to sensor start while scan_ts is a system time.

When you say the system time are you referring to SENSOR_TIME or ROS_TIME?

Samahu, 2024-11-27:
There is probably a flow when using the `TIME_FROM_ROS_TIME` when computing the relative timestamp but the other SENSOR_TIME options should be correct.

I think for the ROS_TIME option we need to do always do `timestamp[ts_idx] - timestamp[0]`

themightyoarfish, 2024-11-28:
Yeah I replaced ROS time with system time for non-ROS use, which seems to be equivalent. I'm now just subtracting the (extrapolated) column 0 timestamp from all others and it seems to be plausible.