ouster-lidar/ouster-ros #423: Destaggered operation and timestamp field in point cloud.

aeudes, 2025-01-23:
After updating the ouster_ros driver from 0.12 to the master:
I struggle to understand what is the correct behaviour of the t field.
From the documentation [here](https://static.ouster.dev/sdk-docs/reference/lidar-scan.html#staggering-and-destaggering) and this issue #199, I expected to have a t constant by columns when in data is in staggered representation and not when in the destagger is apply.
I observe the opposite with the last version of the driver and the validation content of this PR #386 (in link with #350)  seems to be in line with this current observed behaviour.

With the current master of ouster_ros, I observe this. With at the top the time t and bottom the intensity image.
With destagger off:
[image]

With destagger on:
[image]

Could you clarify, if it is the expected behaviour and in this case what is the contents of the t field.

Sensor: OS-0 128
Firmware v3.0.1
ROS : Noetic
Os/machine : Ubuntu 20.04, x64
Git : master

Samahu, 2025-01-23:
Hi @aeudes, when `destagger` flag is set to off this skips step to destagger the point cloud, which means that the PointCloud2 is published in its staggered form that it gets from the sensor. The timestamps associated with each point in the PointCloud would also be staggered in that regard. Meaning every point would still has the right timestamp when this a specific point was recorded by the sensor. That is if you take the staggered **PointCloud2** object and perform destagger operation on it and then dump the intensity and timestamps you should get the same output as if you set the `destagger` flag to true and published a destaggered **PointCloud2** object.

Let me know if this answers your question or find this what I mentioned here contradicts your observation.

aeudes, 2025-01-23:
Thank you for your answer.
I agree with what you say about the destaggering process, my main question here is about the timestamp of points in staggered representation.

from [the documention](https://static.ouster.dev/sdk-docs/reference/lidar-scan.html#staggering-and-destaggering):
```The default representation of LidarScan stores data in staggered columns, meaning that each column contains measurements taken at a single timestamp.```
I expect that if destaggered operation is turn off, the timestamp should be the same in the whole column of the point cloud. And that after the destaggered operation is performed (ordering points with a common azimuth in the same column), then timestamps should be different between points of the same column. And this is not the case in the current data (quite the opposite actually).
My question here : is the timestamp of the point cloud in staggered representation is correct ?

Two possibilities:
- the timestamps is not what I expect or the documentation is misleading.
- the driver miss assign the stamp to the pointcloud in the staggered representation.

P.s. I am in ```TIME_FROM_ROS_TIME``` but I observe this on every frame.

Samahu, 2025-01-24:
@aeudes attached is a dump of the timestamps captured from a staggered frame and a destaggered frame (obtained by switching on and off the `destagger` flag)

[staggered.txt](https://github.com/user-attachments/files/18529110/staggered.txt)
[destaggered.txt](https://github.com/user-attachments/files/18529109/destaggered.txt)

If you take the staggered frame and apply the destagger function to it (which requires the associated [metadata](https://github.com/user-attachments/files/18529054/os-122151001664-metadata.json)) then you will get a frame that is in the same order of timestamps in the destaggered frame.

Hope this clarifies it a bit more

With regard to `TIME_FROM_ROS_TIME` I would advise against using it in the meantime as it suffers from an issue when it comes to relative timestamps.

aeudes, 2025-01-24:
Thanks for the data, it confirm that I observe the same behaviour as in the provided data and I will try to illustrate my point with the data.

I absolutely agree with the fact that the destaggered process correctly matched the time stamp from staggered.txt to destaggered.txt and my question is not on this point.

If we look at the first three column of staggered.txt, the timestamps are:
3108490   3309610   3506950
977150     1161280   1364460 
99017910 99214520 9942342
97075230 97278810 9745496
2932490   3108490   3309610 

This is in contradiction with the documentation on staggered representation that states that:
In staggered representation, points in the same column have the same timestamp:
Could be find here:
from [the documention](https://static.ouster.dev/sdk-docs/reference/lidar-scan.html#staggering-and-destaggering):
> The default representation of LidarScan stores data in staggered columns, meaning that each column contains measurements taken at **a single timestamp**.

or also here:
https://github.com/ouster-lidar/ouster-sdk/blob/69248b4b0c689bf484d201ff1da5172fca1e9003/examples/representations_example.cpp#L117-L121

>  // Fields come in w x h arrays, but they are staggered, **so that a column
    // reflects the timestamp**. To get each column to make visual sense,
    // destagger the image

So, I expect the timestamp in staggered frame to be constant and something like (fictif, no real data,as an illustration):
0 977150 2932490 
0 977150 2932490 
0 977150 2932490 
0 977150 2932490 
0 977150 2932490 

Thanks for your answer and I hope this will clarify my question.

Samahu, 2025-01-24:
I see your point, however, I think the documentation page applies to the un-assembled frame and the above statement is telling you how you should parse the data.

The sensor only send timestamps per packet and not per point, so in the assembled `LidarScan` frame you get one row of timestamps for the different fields in the `LidarScan` .. so if you expand (duplicate) the timestamp row in the `LidarScan` to be a 2D image it will always appear similar to [the upper image in your post](https://private-user-images.githubusercontent.com/462068/406036341-e348dfcf-88f8-4dc1-85c8-c265fdbc2cc8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzc3Mzc5MDQsIm5iZiI6MTczNzczNzYwNCwicGF0aCI6Ii80NjIwNjgvNDA2MDM2MzQxLWUzNDhkZmNmLTg4ZjgtNGRjMS04NWM4LWMyNjVmZGJjMmNjOC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEyNFQxNjUzMjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NDZiOGU0ZDBhYTE3MjM5ZDZiMzRlN2RmMjlmOWE2NmZkZjlmZjNjYmQxOTUyMjMyMWMxMjFlYzk3M2NkNWQ4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.0GkQVQZZxm6JItpa32s0dJgN1LOyEadskx9eO3GTkEU) regardless if you have staggered or destaggered the fields in the `LidarScan`. In this case you would be right about your assumption.

However, when it comes to the `PointCloud2` this is totally different and the `t` timestamp can be open to interpretation. I have to read more about it and check whether there is a standard or consensus on how such field is interpreted. The way we interpret this field is the real timestamp when this point was captured by the sensor. Thus, RANGE and REFLECTIVITY in each point record the actual (relative) time of when this point was captured by the sensor, with this interpretation then the way we associate the time is the correct way. Also consider the case of the non-organized `PointCloud2`. Using this interpretation the user can re-arrange the frame in an orderly manner based on time for certain time sensitive operations such as the `dewarp`.

aeudes, 2025-01-27:
Ok, it makes more sense now.
We have to consider that the timestamp in the column header is related to the measurementid/azimuth but not to the data of fields in the column because they are staggered.
For fields, the destaggered process, reorder measurement to be with correct azimuth. And in the end, the timestamps are the same in each column of the pointcloud.

This was not clear from the documentation and there are no exemple in the sdk of how to recover the timestamps of point (only the timestamp of scan).

As a user, in the point cloud, I expect to have the timestamp related to when the sensor measure the point (accurate for dewarping process). So the current implementation seems to match.

Thank you very much for your help !

Samahu, 2025-01-27:
Absolutely, happy to help!

closed 2025-01-27 by Samahu