ouster-lidar/ouster-ros #446: Release buld for ROS2 branch crashes in configuring stage

lukasj-imar, 2025-03-06:
**Describe the bug**
The actual ros2 branch [8833bee](https://github.com/ouster-lidar/ouster-ros/commit/8833bee80caffabcafe2f7a829fa109827be8dd9) can be built an run in debug mode without any problems. If I switch to a Release build, the os_driver node crashes with a segmentation fault within the configuring stage. I don't know where the crash exactly happens.

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. source ros environment
2. compile the project workspace as release built
3. source the project workspace
4. ros2 launch ouster_ros driver.launch.py

**Debug output:**

[image]

**Platform (please complete the following information):**
- OS1-128 rev 7
- Sensor Firmware: v3.0.1
- ROS2 humble
- Ubuntu 22.04
- arch:x86

Samahu, 2025-03-06:
I don't think I have ever tested the driver against `arch:x86`, Can you see if it is possible to reproduce the issue when using `RelWithDebugInfo` build type?

Samahu, 2025-03-07:
This is likely caused by a package version conflict. Are you a docker user? Would it be possible that you try building the repo provided `Dockerfile` and then use it to attach to the sensor (utilizing the ip address instead of the hostname)? This would help to eliminate the possibility it is a problem associated with the machine and is rather a package problem (the docker container wouldn't have a package problem).

lukasj-imar, 2025-03-12:
Hi again, you're right: It's a dependancy problem. 

With the docker image, the node runs without crashing. I assume the reason could be my prebuilt `pcl-1.14.1` lib.

Thanks for your hint!

closed 2025-03-12 by lukasj-imar