ouster-lidar/ouster-ros #447: Multi OS1-128 LiDAR Initialization Issue on Jetson AGX Orin with ROS2 Humble

nsa05605, 2025-03-10:
I am trying to connect and simultaneously use four OS1-128 LiDAR sensors with a Jetson AGX Orin board. However, I am experiencing an issue where one or two of the LiDARs repeatedly attempt to initialize during runtime. Occasionally, all four LiDARs successfully initialize and run simultaneously, but most of the time, three sensors work correctly while one continuously tries to reinitialize.

I am using ROS2 Humble with the ros2 branch. The launch file I am using is a modified version of driver_launch.py, and each LiDAR has a unique IP address (in the format xxx.xxx.xxx.xx1, etc.) and a different port number. The udp_dest parameter is set to the Jetson PC's IP address. And Lidar Mode is set 1024x10, Timestamp Mode is set TIME_FROM_PTP_1588

```
from pathlib import Path
import launch
from ament_index_python.packages import get_package_share_directory
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription
from launch.substitutions import LaunchConfiguration
from launch.launch_description_sources import PythonLaunchDescriptionSource

def create_driver_launch(sensor_id: int):
    """
    Create an IncludeLaunchDescription action for a specific sensor.
    """
    ouster_ros_pkg_dir = get_package_share_directory('ouster_ros')
    params_file = LaunchConfiguration(f'params_file_{sensor_id}')
    params_file_arg = DeclareLaunchArgument(
        f'params_file_{sensor_id}',
        default_value=str(Path(ouster_ros_pkg_dir) / 'config' / f'os_sensor_cloud_image_params_{sensor_id}.yaml'),
        description=f'Parameters file for Ouster sensor {sensor_id}'
    )
    driver_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([str(Path(ouster_ros_pkg_dir) / 'launch' / 'sensor.composite.launch.py')]),
        launch_arguments={
            'params_file': params_file,
            'ouster_ns': f'ouster{sensor_id}',
            'os_driver_name': f'ouster_driver_{sensor_id}',
            'viz': 'False',
            'rviz_config': './install/ouster_ros/share/ouster_ros/config/community_driver.rviz'
        }.items()
    )
    return params_file_arg, driver_launch

def generate_launch_description():
    """
    Generate launch description for running multiple Ouster ROS sensor nodes.
    """
    sensors = [1, 2, 3, 4]  # List of sensor IDs
    launch_elements = [create_driver_launch(sensor) for sensor in sensors]
    return launch.LaunchDescription([elem for pair in launch_elements for elem in pair])

```

To rule out power issues, I connected the power supplies of each LiDAR to separate power sources, but the problem persisted. When running the launch file, the CPU RAM usage on the Jetson board is approximately 3.3/64 GB, and the power consumption is around 3.2 W.

I also considered bandwidth issues and modified the MTU settings and recv buffer size, but the problem remains.

Is there anything else I should check? Any suggestions would be greatly appreciated.

Samahu, 2025-03-10:
Have you tried to connect to the sensor through a regular laptop?
Is it always the same sensor that you have issue with or it is random?

>  the CPU RAM usage on the Jetson board is approximately 3.3/64 GB

I wasn't able to interpret the CPU RAM, you mean 3.3 GB ram out of 64 available on the board? What about the CPU utilization?

Have you applied the recommended [network config](https://github.com/ouster-lidar/ouster-ros/blob/master/util/network-configure.bash)?
 
Another thing which DDS solution that you are using? CycloneDDS has proven better over time.

Personally, I have only tested connecting 2 sensors at the same time on my , so I don't know if the device is capped by means of CPU or network bandwidth.

nsa05605, 2025-03-10:
I haven’t connected the sensor to a laptop yet. Generally, the third one isn’t working, but sometimes the other ones don’t work either, even when the third one is functioning.

The Jetson AGX Orin has 64 GB of available CPU RAM memory, which is shared between the CPU and GPU. I’ve already checked the GPU memory, and it’s not that high.

I’ve now set rmem_max and rmem_default to 2147483647. If I remember correctly, I’ve also already configured the recv_buffer and mtu. Additionally, I’m currently using CycloneDDS.

If there’s no solution to this issue, could it be a hardware problem? For example, could it be related to the CPU power of the Jetson board or the Ouster sensor’s cable?

Samahu, 2025-03-11:
> CPU RAM

Right but what is the CPU usage?

> If I remember correctly, I’ve also already configured the recv_buffer and mtu.

Note these setting do not persist by default, also the MTU isn't helpful based on our assessment. The RTX buffer configuration might help but I don't think you can set on Jetson (the feature is lacking from tigra linux, at least the one I have played with).

>  For example, could it be related to the CPU power of the Jetson board or the Ouster sensor’s cable?

Definitely, it could be a hardware problem. I for sure had some issues with the Jetson power settings, I have found that higher power setting actually led to more dropped packets.

In terms of Ouster sensors and cabling, you probably want to test every sensor individually and find out if have any problems and if they do then open a support ticket [with our support staff](https://ouster.com/company/contact-us) and they would be happy to assist.

nsa05605, 2025-03-11:
Thank you for your help. I found some error message on Web UI.

The error code is 0x0100003F, 0x01000026, 0x01000027. So I will ask the to support team.

Samahu, 2025-03-12:
Ok, I am going to close this ticket for now, but feel free to re-open if this you think this is rather a driver issue.

closed 2025-03-12 by Samahu

nsa05605, 2025-04-21:
Hello, I’m currently facing a different issue. I’m using 4 Ouster OS1 LiDARs, but the time synchronization between the sensors is not working correctly. Both PTP and phase_lock are enabled on all devices. Do you know any solutions to this problem?

Samahu, 2025-04-21:
It could be something wrong with your PTP configuration, or the sensors are not responding correctly to the PTP master.

I recommend consulting with our guide: https://static.ouster.dev/sensor-docs/image_route1/image_route3/time_sync/time-sync.html#external-trigger-clock-source

nsa05605, 2025-04-22:
I have four Ouster LiDARs installed at 90-degree angles from each other, and I’ve set their phase lock offsets to 0, 90000, 180000, and 270000, respectively. In this configuration, I observe timestamp differences of 0, 25, 50, and 75 ms between the sensors. However, if I set all offsets to 0, the timestamps from all sensors are almost identical.

According to the documentation, the phase lock offset should be configured based on each sensor’s transform. But in that case, the timestamps are no longer synchronized. Is this still considered the correct configuration?

Samahu, 2025-04-22:
I would recommend posting this question to the https://community.ouster.com as it isn't something specific to the ROS driver.