ouster-lidar/ouster-ros #448: Having problems getting the bag file. I want to do COIN-LIO

thomas112200, 2025-03-15:
I want to do [COIN-LIO](https://github.com/ethz-asl/COIN-LIO) with 128 layers sensor of ouster's os-1, but it does not work. The problem is that when I run **driver.launch** on the ROS driver, I get the bag file with **$ rosbag record -a** and play it back while starting COIN-LIO, the point clouds and images are not displayed. I don't know why the bag file doesn't show up well, I checked with the echo command during playback of the bag file and **/ouster/points** is fine, but **/cloud_registered_body**, which is published from COIN-LIO, has no content. Is there anything we should be aware of when doing this in real time and playing back the bag file?

aki662244, 2025-03-19:
Thank you for your reply!
However, I would like to ask because the commands you gave me did not work.
We are currently using these two commands and have changed driver.launch to not start rviz.

`roslaunch ouster_ros driver.launch sensor_hostname:={Link-local IPv4 adress} metadata:={path_to_metadata}.json organized:=false`

`roslaunch coin_lio mapping.launch metadata_file:=${path_to_metadata}json column_shift:=0 point_topic:=/ouster/points imu_topic:=/ouster/imu `

`{path_to_metadata} `specifies a json file obtained from the sensor.
I was able to start COIN-LIO without any problem, but the output image is distorted as shown in the picture. Have you ever seen this kind of phenomenon?
The environment is Noetic with ROS1.

[image]

Also, do you think this warning is the cause of the problem, although I do not think it has anything to do with it?

**[ WARN] [1742361038.670500426]: point_cloud_frame is set to sensor_frame. If you need  to reproject or raytrace the points then use the lidar_frame instead. Refer to: https://github.com/ouster-lidar/ouster-ros/issues/33 for more details.**

Samahu, 2025-03-19:
No the warning has nothing to do with it.

If you go through the COIN-LIO readme, you are supposed to perform sensor calibration first. Have you already performed this step?

Also, I think COIN-LIO package expects Ouster point cloud in its staggered form (during both calibration and mapping step), to disable destaggering append the following option to your launch command `destagger:=false` and re-run calibration and mapping.

aki662244, 2025-03-24:
Hi @Samahu, Sorry for the delay in contacting you.
As it turns out, I was successful in running COIN-LIO.
It was important to set **destagger:=false**. 
Thank you very much for your response to my question！

closed 2025-03-24 by Samahu