ouster-lidar/ouster-ros #451: ouster msgs file path mismatch

kilicdamla, 2025-04-11:
When playing the example ROS bag files, warnings appear due to missing message types. The bag expects PacketMsg.msg and Telemetry.msg under src/ouster-ros/ouster-ros/msg/, but they are actually located in src/ouster-ros/ouster-sensor-msgs/msg/.

Copying the .msg files to the expected directory fixes the issue. It would be helpful to document this or adjust the repo structure.

Samahu, 2025-04-11:
Are you trying to replay a converted ros1 bag file on a the ros2 driver? The `ouster-sensor-msgs` is the path used by ros2 driver. while ros1 driver use `ouster-ros/msg`

kilicdamla, 2025-04-14:
Yes, as you said, I was trying to play a bag file that had been converted from ROS1 to ROS2. So it seems the problems are stemming from that.
In addition, now I’m getting warnings like:
"[WARN] [1744611510.381442828] [rosbag2_player]: Ignoring a topic '/os_node/sys_time', reason: package 'ouster_decoder' not found" and
"[WARN] [1744611510.384009915] [rosbag2_player]: Ignoring a topic '/spot/status/estop', reason: package 'spot_msgs' not found".
I will try to install the relevant packages.

Samahu, 2025-04-14:
I am not sure how you are converting them but here is a PR that I have previously created which provide a starter code for the conversion  https://github.com/ouster-lidar/ouster-ros-extras/pull/2 but i haven't had the time to make it happen. However, it should serve as a good example on matching and renaming the topics. Hope this helps!