ouster-lidar/ouster-ros #452: Failed launch sensor.launch file

Leslie-Jane, 2025-04-13:
> **IMPORTANT**  
> Before reporting a new bug make sure you have:
> - [ ] Checked existing issues https://github.com/ouster-lidar/ouster-ros/issues
> - [ ] Checked existing discussions https://github.com/ouster-lidar/ouster-ros/discussions
> - [ ] Checked ouster community https://community.ouster.com/
> If you couldn't find relevant information and you think this is rather a driver 
> problem then proceed with bug reporting.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. source ros environment
2. compile the project workspace
3. source the project workspace
4. ros launch ouster_ros sensor or replay
5. open another terminal and the project workspace
6. observe the issue ...

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
- Ouster Sensor? \[e.g. OS-0, OS-1, ..\]
- Ouster Firmware Version? \[e.g. 2.3, 2.4, ..\]
- ROS version/distro? \[e.g. noetic, foxy, humble, ..\]
- Operating System? \[e.g. Linux, Windows, MacOS\]
- Machine Architecture? \[e.g. x64, arm\]
- git commit hash (if not the latest).

Leslie-Jane, 2025-04-13:
There was an error or bug when  I launch sensor.launch.
Showing as follow
Ouster ip : 192.168.1.11  local ip: 192.168.1.175
[image]

Leslie-Jane, 2025-05-09:
> It is likely you have some conflicting packages. Can you try to re-produce the issue when running within docker?

It seems not that I have some conflicting packages.Cause I cloned the original ouster driver codes in ros noetic and buildedd、running with roslaunch  ouster_ros driver.launch sensor_hostname:=192.168.1.1,I didn't do anything more.   
The picture showing below is my ouster driver folder constucture.And when I running it in ros2 humble or foxy，things flow smoothly like flowin water.
[image]

Leslie-Jane, 2025-05-09:
> It is likely you have some conflicting packages. Can you try to re-produce the issue when running within docker?

I don't know how to re-produce the issue within docker T_T... I'm SORRY

Samahu, 2025-05-09:
What I mean by package conflict is you may have a specific ros package whether it is PCL or Eigen that doesn't integrate well with the driver. You could either try to re-produce the issue on a fresh system or use docker. To use docker you can refer to the step [here](https://github.com/ouster-lidar/ouster-ros/blob/09c97ebb859327dbe3b7d946ae9103eac7cec110/.github/workflows/docker-image.yml#L24-L28) on building a docker image using the included dockerfile then launch the driver as normal as follow:
```bash
docker run --rm ouster-ros-noetic roslaunch ouster_ros sensor_hostname:=192.168.1.11 viz:=false
```

Samahu, 2025-05-09:
You can also follow the instruction provided [here](https://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) to enable the gdb debugger which should help you pin point the exact exception that is being thrown.

Samahu, 2026-10-22:
Haven't heard back from in a while .. closing the issue.

closed 2026-10-22 by Samahu