ouster-lidar/ouster-ros #453: Getting Segmentation fault when building ouster-ros package for ros2

SyedMIrtazaHyder, 2025-04-16:
> **IMPORTANT**  
> - [x] Checked existing issues https://github.com/ouster-lidar/ouster-ros/issues
> - [x] Checked existing discussions https://github.com/ouster-lidar/ouster-ros/discussions
> - [x] Checked ouster community https://community.ouster.com/
> If you couldn't find relevant information and you think this is rather a driver 
> problem then proceed with bug reporting.

**Describe the bug**

Received the following error message during colcon building phase

```
Starting >>> ouster_sensor_msgs
Starting >>> car_model
Finished <<< car_model [0.68s]
Finished <<< ouster_sensor_msgs [5.96s]
Starting >>> ouster_ros
--- stderr: ouster_ros
/home/user/AV/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/src/lidar_scan.cpp:508:1: internal compiler error: Segmentation fault
  508 | }
        | ^
0xe34ddb internal_error(char const*, ...)
???:0
0x13ac413 ggc_set_mark(void const*)
???:0
0x13a969d gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aae68 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa1d0 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13a9bfb gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aabed gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa494 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aabb2 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13a978d gt_ggc_mx_lang_tree_node(void*)
???:0
0x13a978d gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa5c8 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13b438d gt_ggc_mx_vec_tree_va_gc_(void*)
???:0
0x13b4872 gt_ggc_mx_lang_type(void*)
???:0
0x13aac54 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa4bd gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa014 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa1a8 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13a9bfb gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aabed gt_ggc_mx_lang_tree_node(void*)
???:0
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-11/README.Bugs> for instructions.
gmake[2]: *** [ouster_example/ouster_client/CMakeFiles/ouster_client.dir/build.make:132: ouster_example/ouster_client/CMakeFiles/ouster_client.dir/src/lidar_scan.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
/home/user/AV/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/src/sensor_info.cpp: In function ‘ouster::sensor::sensor_info ouster::sensor::metadata_from_json(const string&, bool)’:
/home/user/AV/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/src/sensor_info.cpp:976:1: internal compiler error: Segmentation fault
976 | }
| ^
0xe34ddb internal_error(char const*, ...)
???:0
0x13ac413 ggc_set_mark(void const*)
???:0
0x13a969d gt_ggc_mx_lang_tree_node(void*)
???:0
0x13a9f70 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa576 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13b438d gt_ggc_mx_vec_tree_va_gc_(void*)
???:0
0x13b4872 gt_ggc_mx_lang_type(void*)
???:0
0x13aac54 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa4bd gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa014 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa494 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa014 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13a9fda gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa494 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13b438d gt_ggc_mx_vec_tree_va_gc_(void*)
???:0
0x13b4872 gt_ggc_mx_lang_type(void*)
???:0
0x13aac54 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aa4bd gt_ggc_mx_lang_tree_node(void*)
???:0
0x13aabb2 gt_ggc_mx_lang_tree_node(void*)
???:0
0x13a9f36 gt_ggc_mx_lang_tree_node(void*)
???:0
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-11/README.Bugs> for instructions.
gmake[2]: *** [ouster_example/ouster_client/CMakeFiles/ouster_client.dir/build.make:104: ouster_example/ouster_client/CMakeFiles/ouster_client.dir/src/sensor_info.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:589: ouster_example/ouster_client/CMakeFiles/ouster_client.dir/all] Error 2
gmake: *** [Makefile:166: all] Error 2
---
Failed   <<< ouster_ros [13.6s, exited with code 2]

Summary: 2 packages finished [19.7s]
1 package failed: ouster_ros
1 package had stderr output: ouster_ros
```

**To Reproduce**
1. make container with image
2. source ros2 workspace
3. in the workspace src folder run: `git clone -b ros2 --recurse-submodules https://github.com/ouster-lidar/ouster-ros.git`
4. Go back to workspace folder and then run `colcon build --from-path src -y --ignore-src -DMAKE_BUILD_TYPE=Release`
5. Gives Segfault

**Platform (please complete the following information):**
- Ouster Sensor? OS1-Gen1
- Ouster Firmware Version? 2.1
- ROS version/distro? Humble
- Operating System? Ubuntu Noble 2024.04, but running ros in Docker container (osrf/ros:humble-desktop-full)[https://hub.docker.com/r/osrf/ros/tags]
- Machine Architecture? AMD Ryzen 5 4500U with Radeon Graphics
- git commit hash: latest and tags/ros2-v0.13.2

Samahu, 2025-04-16:
This seem like a compiler problem, try turning all of the follow:
* Turn off code optimization `g++ -O0` or `gcc -O0`
* Change the current gcc version (so consider downgrading to 9/10 or 12)
* Switch to clang if possible

SyedMIrtazaHyder, 2025-04-17:
1. Used flag `-DCMAKE_BUILD_TYPE=Debug`, which should remove any code optimizations in the system.
2. Switched to clang 14 to get another segmentation fault:
```
--- stderr: ouster_ros
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace, preprocessed source, and associated run script.
Stack dump:
0.      Program arguments: /usr/bin/clang++ -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DEIGEN_MPL2_ONLY -DRCUTILS_ENABLE_FAULT_INJECTION -I/home/user/AV/src/ouster-ros/ouster-ros/include -I/home/user/AV/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/include -I/home/user/AV/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/include/optional-lite -I/home/user/AV/src/ouster-ros/ouster-ros/ouster-sdk/ouster_pcap/include -isystem /opt/ros/humble/include/rclcpp -isystem /opt/ros/humble/include/class_loader -isystem /opt/ros/humble/include/rclcpp_components -isystem /opt/ros/humble/include/ament_index_cpp -isystem /opt/ros/humble/include/libstatistics_collector -isystem /opt/ros/humble/include/builtin_interfaces -isystem /opt/ros/humble/include/rosidl_runtime_c -isystem /opt/ros/humble/include/rcutils -isystem /opt/ros/humble/include/rosidl_typesupport_interface -isystem /opt/ros/humble/include/fastcdr -isystem /opt/ros/humble/include/rosidl_runtime_cpp -isystem /opt/ros/humble/include/rosidl_typesupport_fastrtps_cpp -isystem /opt/ros/humble/include/rmw -isystem /opt/ros/humble/include/rosidl_typesupport_fastrtps_c -isystem /opt/ros/humble/include/rosidl_typesupport_introspection_c -isystem /opt/ros/humble/include/rosidl_typesupport_introspection_cpp -isystem /opt/ros/humble/include/rcl -isystem /opt/ros/humble/include/rcl_interfaces -isystem /opt/ros/humble/include/rcl_logging_interface -isystem /opt/ros/humble/include/rcl_yaml_param_parser -isystem /opt/ros/humble/include/libyaml_vendor -isystem /opt/ros/humble/include/tracetools -isystem /opt/ros/humble/include/rcpputils -isystem /opt/ros/humble/include/statistics_msgs -isystem /opt/ros/humble/include/rosgraph_msgs -isystem /opt/ros/humble/include/rosidl_typesupport_cpp -isystem /opt/ros/humble/include/rosidl_typesupport_c -isystem /opt/ros/humble/include/composition_interfaces -g -fPIE -Wall -Wextra -std=gnu++17 -MD -MT CMakeFiles/os_cloud.dir/rclcpp_components/node_main_os_cloud.cpp.o -MF CMakeFiles/os_cloud.dir/rclcpp_components/node_main_os_cloud.cpp.o.d -o CMakeFiles/os_cloud.dir/rclcpp_components/node_main_os_cloud.cpp.o -c /home/user/AV/build/ouster_ros/rclcpp_components/node_main_os_cloud.cpp
1.      /opt/ros/humble/include/rclcpp/rclcpp/parameter_client.hpp:305:3 <Spelling=/opt/ros/humble/include/rclcpp/rclcpp/macros.hpp:102:3>: current parser token '{'
2.      /opt/ros/humble/include/rclcpp/rclcpp/parameter_client.hpp:46:1: parsing namespace 'rclcpp'
3.      /opt/ros/humble/include/rclcpp/rclcpp/parameter_client.hpp:302:1: parsing struct/union/class body 'rclcpp::SyncParametersClient'
4.      /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/unique_ptr.h:242:11: instantiating class definition 'std::unique_ptr<rclcpp::SyncParametersClient>'
5.      /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/unique_ptr.h:208:12: instantiating class definition 'std::__uniq_ptr_data<rclcpp::SyncParametersClient, std::default_delete<rclcpp::SyncParametersClient>, true, true>'
6.      /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/unique_ptr.h:128:11: instantiating class definition 'std::__uniq_ptr_impl<rclcpp::SyncParametersClient, std::default_delete<rclcpp::SyncParametersClient>>'
7.      /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/tuple:609:11: instantiating class definition 'std::tuple<rclcpp::SyncParametersClient *, std::default_delete<rclcpp::SyncParametersClient>>'
8.      /usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/tuple:250:12: instantiating class definition 'std::_Tuple_impl<0, rclcpp::SyncParametersClient *, std::default_delete<rclcpp::SyncParametersClient>>'
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
/lib/x86_64-linux-gnu/libLLVM-14.so.1(_ZN4llvm3sys15PrintStackTraceERNS_11raw_ostreamEi+0x31)[0x7b9c6720ad01]
/lib/x86_64-linux-gnu/libLLVM-14.so.1(_ZN4llvm3sys17RunSignalHandlersEv+0xee)[0x7b9c67208a3e]
/lib/x86_64-linux-gnu/libLLVM-14.so.1(_ZN4llvm3sys15CleanupOnSignalEm+0xfb)[0x7b9c6720a0ab]
/lib/x86_64-linux-gnu/libLLVM-14.so.1(+0xd6bdff)[0x7b9c67136dff]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7b9c65eaf520]
/lib/x86_64-linux-gnu/libclang-cpp.so.14(+0x175b2a9)[0x7b9c6e3f82a9]
/lib/x86_64-linux-gnu/libclang-cpp.so.14(+0x1754356)[0x7b9c6e3f1356]
/lib/x86_64-linux-gnu/libclang-cpp.so.14(+0x174e695)[0x7b9c6e3eb695]
/lib/x86_64-linux-gnu/libclang-cpp.so.14(_ZN5clang4Sema23DeduceTemplateArgumentsEPNS_20FunctionTemplateDeclEPNS_24TemplateArgumentListInfoEN4llvm8ArrayRefIPNS_4ExprEEERPNS_12FunctionDeclERNS_4sema21TemplateDeductionInfoEbNS5_12function_refIFbNS6_INS_8QualTypeEEEEEE+0x6bb)[0x7b9c6e3eaf9b]
/lib/x86_64-linux-gnu/libclang-cpp.so.14(_ZN5clang4Sema28AddTemplateOverloadCandidateEPNS_20FunctionTemplateDeclENS_14DeclAccessPairEPNS_24TemplateArgumentListInfoEN4llvm8ArrayRefIPNS_4ExprEEERNS_20OverloadCandidateSetEbbbNS_8CallExpr11ADLCallKindENS_27OverloadCandidateParamOrderE+0x2b3)[0x7b9c6e2e8823]
[... cut ...]
clang: note: diagnostic msg: /tmp/node_main_os_cloud-4f2a7d.cpp
clang: note: diagnostic msg: /tmp/node_main_os_cloud-4f2a7d.sh
clang: note: diagnostic msg:

********************
gmake[2]: *** [CMakeFiles/os_cloud.dir/build.make:76: CMakeFiles/os_cloud.dir/rclcpp_components/node_main_os_cloud.cpp.o] Error 139
gmake[1]: *** [CMakeFiles/Makefile2:399: CMakeFiles/os_cloud.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:166: all] Error 2
---
Failed   <<< ouster_ros [6.95s, exited with code 2]
```

SyedMIrtazaHyder, 2025-04-17:
For now I have just spammed the colcon build command and it seemed to have built it. It always went up a percentage and when it was built it also started to work.
Any idea why this was happening inside the container?

Samahu, 2025-04-17:
I am not exactly sure ... this could occur if you have a limited amount of RAM below 32GB or if you something wrong with the system .. but since you are using docker the OS shouldn't be a problem. So just guessing I would say you have insufficient RAM.

Samahu, 2025-04-23:
@SyedMIrtazaHyder I haven't heard back from you on this. As I noted this is likely due to limited memory available on the machine you are using to build the driver, but should be easy to solve by disabling parallel build. i.e. try to build with: `colcon build --executor sequential ... ` and see if this help. I am going to close this ticket but let us know if u still face issue with the driver.

closed 2025-04-23 by Samahu