ouster-lidar/ouster-ros #454: os_sensor does not reject invalid packets

Aposhian, 2025-04-17:
**Describe the bug**
At least in my configuration, with IPv4 and no mulitcast, `udp_data_socket` binds 0.0.0.0, so then it accepts any data sent on port 7502 on any of the computers interfaces. The situation I ran into was unicast DDS discovery messages being sent to 7502 as well, but on loopback (according to this https://cyclonedds.io/docs/cyclonedds/latest/config/port_numbers.html). When these discovery messages showed up every 8 seconds, the `os_sensor` node would log:

```
Unexpected udp packet length: 544
```
and then output garbage data.

Obviously, it is up to the user to make sure that ports don't conflict, but I think that the Ouster SDK could make this easier to troubleshoot by:
- binding only to the address/interface that the lidar is actually on to limit the amount of conflict (This is related to the Ouster SDK: https://github.com/ouster-lidar/ouster-sdk/issues/653)
- more clearly log when packets are completely invalid

I think that if `read_lidar_packet` returns false (didn't read a whole packet), then `write_overwrite` should not advance the circular buffer, we should force those packets from not being published.

I think it probably makes this issue more tricky to track down, because if you send a smaller UDP packet than the lidar packets to the port, then it will only overwrite a portion of the buffer, and `write_overwrite` will still just advance by an item size, and so what is left in the buffer is mostly stale data from a previous packet, but with part of it as garbage.

**To Reproduce**
- Setup data to stream on port 7502 (default)
- Send UDP packets to port 7502 on loopback/localhost

**Platform (please complete the following information):**
- Ouster Sensor? OS-0
- Ouster Firmware Version? 2.5
- Programming Language? C++
- Operating System? Linux
- Machine Architecture? x64
- git commit hash? 483206f0baeffbbc11ee0883d54a651a5216563b

Samahu, 2025-04-18:
Yeah I don't think currently we reject the packet if it doesn't match the expected packet size. I think the driver is setup to warn if such packets exist and after a definitive number of errors reached the driver assumes there is a fault and attempts to reset the sensor. It is not a very common scenario that users run into if they have configured their network and other apps correctly.

Aposhian, 2025-05-13:
> I think the driver is setup to warn if such packets exist and after a definitive number of errors reached the driver assumes there is a fault and attempts to reset the sensor.

The problem is that as soon as the driver gets one good packet, it resets the fault count, so if the issue is interspersed with valid data then it never resets.