ouster-lidar/ouster-ros #455: Error when executing COIN-LIO

aki662244, 2025-04-18:
I will share with you a problem I am having with running COIN-LIO by starting driver with OS-1-128. Currently, we are running into a problem where COIN-LIO is almost impossible to process. Specifically, it takes a long time to generate the /tf, camere_init, and the update frequency is extremely slow. I wish it were a simple PC processing power issue, but since it was working fine at first, I do not believe it is the only problem.
As far as what we have investigated, ` rostopic list` and `rostopic hz /ouster/points` and `rostopic hz /ouster/imu` are fine. Also, when I first retrieved the bag file, the timestamp mode was the default, but now I set it to TIME_FROM_ROS_TIME and there was no change. bag was retrieved using a laptop, and on each of those laptops and desktops, bag was The results did not change at all.

What could be the cause other than the timestamp and PC specs?

aki662244, 2025-04-20:
I apologize for the lack of clarity in my question. It was unexpected and I am still trying to figure out the cause. I would like to ask a question about something I have researched and tried by myself and am wondering about.
As for time synchronization by ptp, the laptop I am using does not support ptp. But if the purpose is to get the bag, there is no need to synchronize the PC and sensor, as long as the time of LiDAR and IMU are the same, right?

Samahu, 2025-04-23:
> But if the purpose is to get the bag, there is no need to synchronize the PC and sensor, as long as the time of LiDAR and IMU are the same, right?

Yes, as long as you don't need to sync with a different sensors ..say an external imu then it wouldn't make a difference whatever sensor timestamp mode you planning to use..

Samahu, 2025-04-23:
If you do need to sync with an external sensors then you need to either have hardware based sync mechanism or have all participating sensors be subscribing to a single PTP master, a third option would be to use the TIME_FROM_ROS_TIME and rely solely on the ROS clock for sync.

closed 2025-04-25 by Samahu