ouster-lidar/ouster-ros #458: frame rate issue

JereKnuutinen, 2025-05-21:
Hi,

I changed the fast dds to cyclone dds using "export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp". I also tuned "sudo sysctl -w net.core.rmem_max=2147483647". then i launched driver using ros2 launch ouster_ros driver.launch.py. My issue is that for some reason point cloud visualization in rviz2 is very lagging. However I run in separate terminal "ros2 topic hz /ouster/points" the rviz in not anymore lagging behind. Why is that?
I am using ros2 jazzy and ubuntu 24.04.

Samahu, 2025-05-21:
Some users reported differences in behavior depending on the number of point cloud topic subscribers but usually their complain that more subscribers result in reduced frame rate. We did address that a while back. But this first time seeing that more subscribers fix the lagging in rviz. It could be due to QoS setting between RVIZ and the driver. Have you tried running without rviz and check the frame rate you are getting through `ros2 topic hz`?

jake3991, 2025-06-05:
Update on my side, I was only able to resovle the RVIZ latency AND topic latecy when using cyclone DDS and the following 

```
sudo sysctl -w net.core.rmem_max=16777216    # 16MB max receive buffer
sudo sysctl -w net.core.wmem_max=16777216    # 16MB max send buffer
sudo sysctl -w net.core.rmem_default=1048576 # 1MB default receive buffer
sudo sysctl -w net.core.wmem_default=1048576 # 1MB default send buffer

```