ouster-lidar/ouster-ros #465: ouster-sdk submodule pointing to non-existent ref

ovaag, 2025-06-20:
Current master is pointing to ouster-sdk with hash [d996768](https://github.com/ouster-lidar/ouster-sdk/commit/d99676858b3b6f90e1618354ff7930cabafa9769)

However I'm not able to find this commit with git log after cloning the submodule. 

Also getting this warning when clicking on the submodule link from ouster-ros master
[image]

Should it instead point to this commit [483206f](https://github.com/ouster-lidar/ouster-sdk/commit/483206f0baeffbbc11ee0883d54a651a5216563b) which has the same commit msg (Release 0.11.1), but is a commit on ouster-sdk master branch ?

vsood257, 2025-07-08:
Facing the same issue. [483206f](https://github.com/ouster-lidar/ouster-sdk/commit/483206f0baeffbbc11ee0883d54a651a5216563b) does work. 

Also do we plan to upgrade ouster-sdk to something recent like 0.14.0?

Samahu, 2025-07-08:
Hi @ovaag, the commit does exist. Possibly this was a glitch in GitHub that couldn't serve the commit?

vsood257, 2025-07-09:
I didn’t have any particular feature in mind. I was just surprised to see that we were already at version 0.11.1. After looking through the changelog, though, I didn’t find any features that would be relevant or useful for me. Thanks for the response!

Samahu, 2026-10-22:
I believe this issue has been resolved .. I am also in the process of upgrading to sdk 1.0.1 here:  https://github.com/ouster-lidar/ouster-ros/pull/583

closed 2026-10-22 by Samahu